rerun-io/rerun #8056: Refactor viewport blueprint logic

emilk, 2024-11-11:
The viewport blueprint code is a mess, and modifying it is extremely error prone

* This is blocking #3135 
* Related: https://github.com/rerun-io/rerun/issues/4687

No one is to blame - this has evolved over time - but we need to untangle this.

## How and why is it messy?
I don't fully understand all of it.

### Triplicate state
* As components in the blueprint store (persisted)
* As `egui_tiles::Tree` (created at the start of each frame)
* As `struct ViewportBlueprint` (created at the start of each frame)

### Existence of blueprint entities is significant
We use `SystemCommand::DropEntity` to permanently remove space-view entities from the blueprint store. This means we cannot undo the removal of a space-view.

Our loader should instead start with the `RootContainer` and recuse down, and ignore all entities that aren't reachable.
When I tried to implement this however, I created more trouble for myself.

### Views
We view the `ViewportBlueprint` it in two places:
* The `BlueprintTree` panel (left)
* The viewport panel (center)

If we only mutated the state at the end/beginning of the frame, this wouldn't be a problem,
but I suspect we actually mutate state _between_ calling these two views, which of course causes problems.

### Auto-layout
We want to automatically create space-views if there is no explicit blueprint, leading to complicated state like `auto_layout` and `auto_space_views`:
https://github.com/rerun-io/rerun/blob/31a166f48e89626184e8800ab749b123154c6b31/crates/viewer/re_viewport_blueprint/src/viewport_blueprint.rs#L44-L54

### Delayed action
Almost all edits of the state is delayed, often in multiple steps.

This is because of (at least) two reasons (as far as I can tell)
* we cannot modify a `egui_tiles::Tree` while we're iterating over it
* when modifying the state, we want to modify all THREE copies of it… and the best way to do that is to wait until the end of the frame so we only edit the persisted store state

Delays happen in
* [`TreeAction`](https://github.com/rerun-io/rerun/blob/31a166f48e89626184e8800ab749b123154c6b31/crates/viewer/re_viewport_blueprint/src/tree_actions.rs)
* A `TreeAction` can however lead to another frame-delay: https://github.com/rerun-io/rerun/blob/31a166f48e89626184e8800ab749b123154c6b31/crates/viewer/re_viewport/src/viewport.rs#L415-L421

abey79, 2024-11-19:
GUI automated testing would require `TreeActions` to be somehow processed by our `TestContext` mock. The mess described in this issue makes this basically impossible. A refactor of the above would ideally lead to some `execute_tree_actions` function in `re_viewport_blueprint` that could be called by `TestContext` as appropriate. (Currently this function lives in `re_viewport` and its current implementation is deeply entangled with egui_tile stuff.)

cc @lucasmerlin

emilk, 2024-11-22:
## Plan

* We load the `egui_tioles::Tree` from the store at the start of the frame, recursing from root container
* If there is no root, or if it is pointing to an empty container, this means the viewport is and should be empty
* When we add space-views, they need to be inserted in the tree (somewhere)
* When running auto-layout, we should get all space-views currently in the tree and run auto-layout on that
* Auto-layout is run at the END of the frame
* As the LAST step of the frame, we write the tree to the store (if modified)

emilk, 2024-11-27:
Plan:
* [ ] Put `Viewport` into `ViewerContext` (for less parameter passing everywhere)?
* [ ] Switch the `AtomicBools` for `TreeCommand`
* [ ] Make `Viewport` the mutable version of `ViewportBlueprint` (i.e. put the command queue in Viewport)
* [ ] Only write to the blueprint store inside `update_and_sync_tile_tree_to_blueprint`

closed 2024-11-29 by emilk