rerun-io/rerun #8109: Buggy maps view with time range

nikolausWest, 2024-11-13:
Tested on `rerun-sdk 0.20.0a3`. See video (nuscenes example)

https://github.com/user-attachments/assets/6a58b8be-6fc0-4e52-9bb8-1407515c8224

abey79, 2024-11-13:
There are several things at play there:

1. The "jumping to default" position (aka Rerun HQ) whenever no data is in the view. That's very annoying and I'm going to fix it asap. (Note: with this example, there are two ways to have no data: move the cursor all the way to the left; and set an empty-interval time range.)
2. The "set zoom level on double click" behaviour.

To expend on the latter, double click does two things:
- engage "follow" mode (aka auto-scroll to track the object or mean position of all objects)
- set the zoom level to fit the data (that's one-shot, not continuously updated like the scrolling)

There are technical reasons for why things are that way. There are certainly ways to improve on that behaviour, but that requires more work than can fit for 0.20.

abey79, 2024-11-13:
Related:
- https://github.com/rerun-io/rerun/issues/7884

abey79, 2024-12-17:
Can this be closed or is there more to do? (Cursory testing seems to indicate working as intended now.)

closed 2024-12-17 by nikolausWest