rerun-io/rerun #8125: Monochrome images display range and LUT

Famok, 2024-11-13:
**Describe the solution you'd like**
For monochrome images of type uint8, uint16 and float32 it is often desirable to
1. set the range displayed (best would be min and max value of the source image)
2. display them in "color" using a LUT
3. have an optional auto-contrast (e.g. like Fiji / ImageJ has), which is then either image by image or in case of videos globally set

**Describe alternatives you've considered**
In "older" versions i could achieve this setting the display type as tensor, which somehow allows these features. Now I can use the depth image visualizer, which has an auto-range which is not always wanted.

Wumpf, 2024-11-13:
Very closely related to
* https://github.com/rerun-io/rerun/issues/4624
* https://github.com/rerun-io/rerun/issues/6897

(keeping this open though since imho it provides a different perspective on the problem/feature request!)

Famok, 2024-11-13:
Found another one:
https://github.com/rerun-io/rerun/issues/6976

Famok, 2025-01-10:
I started using depth image to get the same result, which allows setting a LUT and a min an max value. The default behaviour seems to be an auto contrast (for each image individually). Could you just copy paste these functions to the "standard" image?