rerun-io/rerun #8147: Nesting Horizontal containers inside each others via dragging should be deactivated

grtlr, 2024-11-14:
**Describe the bug**

Nesting _Horizontal_ or _Vertical_ containers should be disallowed in the UI. This works for the _Add_ button by greying out these options. However, when dragging a horizontal container inside another horizontal container it simply disappears without user feedback.

As per @Wumpf, this might be related to the automatic cleanup of containers being to aggresive.

**To Reproduce**

https://github.com/user-attachments/assets/36b4696d-468a-4b8f-a2a6-99e46ffd961c

**Expected behavior**
The dragging target should be dissallowed, or the user should be informed about the cleanup.

**Desktop (please complete the following information):**
 - OS: macOS Ventura 13.7

**Rerun version**
Pre-release `release-0.20.0`

Wumpf, 2024-12-04:
@abey79 any relationship to ongoing drag & drop refinement?

abey79, 2024-12-05:
indirectly. We should disallow this. Providing feedback is tricky though.

grtlr, 2025-04-08:
I just checked again and think the current behavior on `main` works well.

closed 2025-04-08 by grtlr