rerun-io/rerun #8165: regression: docs.rs no longer showing docs for rerun 0.20

astraw, 2024-11-17:
**Describe the bug**

https://docs.rs/ no longer builds for rerun 0.20.0, but this [worked with 0.19.1](https://docs.rs/rerun/0.19.1/rerun/).

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://docs.rs/crate/rerun/latest
2. Observe `docs.rs failed to build rerun-0.20.0`

**Expected behavior**
1. Go to https://docs.rs/crate/rerun/latest
2. Observe correctly built docs

**Rerun version**
0.20.0

**Additional context**
I thought this was a spurious network error, but actually it is by design that docs.rs does not allow network access. This is described in the following issue: https://github.com/rust-lang/docs.rs/issues/2674#issuecomment-2481076023

Wumpf, 2024-11-17:
We totally missed that! Thank you so much for filing and already checking with rust-long/docs.rs - will fix the offending build script and publish a patch soon.

Will need to figure out how to test our docs in an isolated environment like that as well!

Wumpf, 2024-11-17:
Meanwhile, we also host our the docs here:
https://ref.rerun.io/docs/rust/head/rerun/

I'll update the link on our docs to point to that temporarily

Wumpf, 2024-11-21:
Fixed in 0.20.1

closed 2024-11-21 by Wumpf

astraw, 2024-11-29:
So this happened again in 0.20.2: https://docs.rs/crate/rerun/latest . I don't seem to be able to re-open the issue.

teh-cmc, 2024-11-29:
The docs.rs worker timed out...
```
[ERROR] error running command: command timed out after 1800 seconds
```

Thanks for the heads up @astraw -- I'll open an issue (there's nothing we can do on our hand though, besides reaching out to them).

astraw, 2024-12-02:
So it looks the build https://docs.rs/crate/rerun/0.20.2/builds/1569209 passed, used rustc 1.84.0-nightly (3fee0f12e 2024-11-20), and took 24m 50s. On the other hand, the build https://docs.rs/crate/rerun/0.20.2/builds/1567546 failed, used rustc 1.85.0-nightly (6b6a867ae 2024-11-27), and timed out after 1800 seconds, as noted by @teh-cmc.

I'm not sure why docs.rs does not use the successful build, but indeed this doesn't seem like anything rerun can fix by itself.

astraw, 2024-12-02:
Oh, I think my browser may have cached a stale version. It looks like 0.20.2 docs are online now. https://docs.rs/rerun/0.20.2/rerun/index.html