rerun-io/rerun #8167: Error with `log_file_from_path` when specifying recording

EtaLoop, 2024-11-18:
With this code :
```
rec = rr.new_recording(application_id="My window", recording_id="My rec", spawn=True)
rr.log_file_from_path(file_path="file.stl", recording=rec)
```

I get the following error : 
```
RerunWarning: log_file_from_path: TypeError(argument 'recording': 'RecordingStream' object cannot be converted to 'PyRecordingStream')
  rr.log_file_from_path(file_path="file.stl", recording=rec)
```
Did I miss something with `log_file_from_path` ?

qianyizhang, 2024-11-20:
> There's a small type definition bug in the Python SDK.
> 
> Until we ship a fix, the workaround is simply to call `RecordingStream.to_native()`:
> 
> ```diff
> rec = rr.new_recording(application_id="My window", recording_id="My rec", spawn=True)
> - rr.log_file_from_path(file_path="file.stl", recording=rec)
> + rr.log_file_from_path(file_path="file.stl", recording=rec.to_native())
> ```

this still doesn't log to desired recording in case of serve then connect.

qianyizhang, 2024-11-20:
to me more specific:
1. `rerun --serve` in cli
2. make some local rrds
```
import rerun as rr
NUM_CASE = 3
path_list = []
for index in range(1, 1+NUM_CASE):
    case_id = f"case{index}"
    rr.init("demo", recording_id=case_id)
    local_path = f"./{case_id}.rrd"
    rr.save(local_path)
    path_list.append(local_path)
    rr.log(
        "triangle{case_id}",
        rr.Mesh3D(
            vertex_positions=[[0.0, index, 0.0], [index, 0.0, 0.0], [0.0, 0.0, 0.0]],
            vertex_normals=[0.0, 0.0, 1.0],
            vertex_colors=[[0, 0, 255], [0, 255, 0], [255, 0, 0]],
            triangle_indices=[2, 1, 0],
        ),
    )
```
3. natively log_file_from_path works as expected
```
import rerun as rr
rr.init("demo", recording_id="does not matter")
rr.connect()
for path in path_list:
    rr.log_file_from_path(path) # log to pre-saved recordings, eg. case1-3
```
4. log_file_from_path with specified recording doesn't work
```
import rerun as rr
rr.init("demo", recording_id="new_record")
rr.connect()
new_recording = rr.get_global_data_recording()
print(new_recording .get_recording_id()) # new_record
for path in path_list:
    rr.log_file_from_path(path, recording=new_recording.to_native()) # expect to add this to current recording, but still ends up in pre-saved recordings, eg. case1-3
```

teh-cmc, 2024-11-20:
:+1: Let's create a separate issue for this:
* #8185

EtaLoop, 2024-11-20:
Thanks, this works now !

closed 2024-11-25 by jleibs