rerun-io/rerun #8190: Unable to compile Rust `rerun` dependency due to re_arrow2

simon-wh, 2024-11-20:
I've been trying to update to rerun v0.20 (from v0.15), but I keep getting compilation errors in the `re_arrow2` dependency.

```
error[E0004]: non-exhaustive patterns: `arrow_schema::DataType::BinaryView`, `arrow_schema::DataType::Utf8View`, `arrow_schema::DataType::ListView(_)` and 1 more not covered
   --> /Users/USER/.cargo/registry/src/index.crates.io-6f17d22bba15001f/re_arrow2-0.17.4/src/datatypes/mod.rs:286:15
    |
286 |         match value {
    |               ^^^^^ patterns `arrow_schema::DataType::BinaryView`, `arrow_schema::DataType::Utf8View`, `arrow_schema::DataType::ListView(_)` and 1 more not covered
    |
note: `arrow_schema::DataType` defined here
   --> /Users/USER/.cargo/registry/src/index.crates.io-6f17d22bba15001f/arrow-schema-53.2.0/src/datatype.rs:97:1
    |
97  | pub enum DataType {
    | ^^^^^^^^^^^^^^^^^
...
270 |     BinaryView,
    |     ---------- not covered
...
287 |     Utf8View,
    |     -------- not covered
...
300 |     ListView(FieldRef),
    |     -------- not covered
...
315 |     LargeListView(FieldRef),
    |     ------------- not covered
    = note: the matched value is of type `arrow_schema::DataType`
help: ensure that all possible cases are being handled by adding a match arm with a wildcard pattern as shown, or multiple match arms
    |
340 ~             DataType::RunEndEncoded(_, _) => panic!("Run-end encoding not supported by arrow2"),
341 ~             _ => todo!(),
    |

    Checking re_memory v0.20.1-alpha.3
    Checking dirs-next v2.0.0
For more information about this error, try `rustc --explain E0004`.
error: could not compile `re_arrow2` (lib) due to 1 previous error
warning: build failed, waiting for other jobs to finish...
```

I was trying to see if I could downgrade the `arrow` dependencies to v52 based to see if it was just grabbing too new of a version, but there are other rerun dependencies that have a strict requirement to v53 based.

OS: MacOS Sonoma 14.5 (using Apple silicon mac)
```
active toolchain
----------------

stable-aarch64-apple-darwin (default)
rustc 1.82.0 (f6e511eec 2024-10-15)
```

simon-wh, 2024-11-20:
When I downgrade to `rerun = "0.19"` then I'm able to compile successfully

emilk, 2024-11-21:
Does `cargo install rerun-cli` work for you?

Are you pulling in `arrow` and/or `re_arrow2` as a dependency yourself? If so, what version?

emilk, 2024-11-21:
Thanks for the reply (and kind words)! I'm gonna chalk this up to cargo being weird then, and closing this issue.

We're moving away from re_arrow2 pretty soon anyway :]

closed 2024-11-21 by emilk