rerun-io/rerun #8198: Help text does not show up

Wumpf, 2024-11-21:
Those little `?` seem to be broken.

[image]

@abey79 says this might be due to the egui update
Doesn't seem to happen on 0.20

abey79, 2024-11-21:
Well I can't reproduce on `main` as of now 🤷🏻

Wumpf, 2024-11-22:
That is for all views. But I noticed now that this doesn't repro so far anywhere except the following rrd + rbl: https://drive.google.com/file/d/13LqtoFHFbwrqESvszptPXXLeOyUP7UQy/view?usp=sharing

abey79, 2024-11-25:
Can't repro with this file either:

[image]

closed 2025-01-31 by emilk