rerun-io/rerun #8231: version 0.20.1 fails to build in non-cargo projects

idobenamram, 2024-11-27:
**Describe the bug**
It looks like in rerun 0.20.0-0.20.1 `re_build_tools` started using `cargo metadata`. we compile our rust code cargo using bazel (crates universe) which means cargo metadata doesn't work for us. What do you guys suggest we do?

```
Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
Build script process failed with exit code 101
--stdout:
cargo:rerun-if-env-changed=IS_IN_RERUN_WORKSPACE
cargo:rerun-if-env-changed=RERUN_IS_PUBLISHING_CRATES
cargo:rustc-env=RE_BUILD_DATETIME=
cargo:rustc-env=RE_BUILD_GIT_HASH=
cargo:rustc-env=RE_BUILD_GIT_BRANCH=
cargo:rustc-env=RE_BUILD_TARGET_TRIPLE=x86_64-unknown-linux-gnu
cargo:rustc-env=RE_BUILD_RUSTC_VERSION=1.82.0 (f6e511eec 2024-10-15)
cargo:rustc-env=RE_BUILD_LLVM_VERSION=19.1.1
cargo:rustc-env=RE_BUILD_IS_IN_RERUN_WORKSPACE=
cargo:rerun-if-env-changed=IS_IN_RERUN_WORKSPACE
cargo:rerun-if-env-changed=RERUN_IS_PUBLISHING_CRATES

--stderr:
Environment: Not on CI and not in workspace
Environment: Not on CI and not in workspace
thread 'main' panicked at external/_main~rust_crates~crate_index__re_build_tools-0.20.1/src/lib.rs:204:46:
called `Result::unwrap()` on an `Err` value: failed to start `cargo metadata`: No such file or directory (os error 2)

Caused by:
    No such file or directory (os error 2)
```

**Backtrace**
```
Stack backtrace:
   0: <unknown>
   1: <unknown>
   2: <unknown>
   3: <unknown>
   4: <unknown>
   5: <unknown>
   6: <unknown>
   7: <unknown>
   8: <unknown>
   9: __libc_start_main
  10: <unknown>

stack backtrace:
   0:     0x6526fe7ee4fa - <unknown>
   1:     0x6526fe8148cb - <unknown>
   2:     0x6526fe7eb2f3 - <unknown>
   3:     0x6526fe7ef842 - <unknown>
   4:     0x6526fe7ef4ae - <unknown>
   5:     0x6526fe7f00ef - <unknown>
   6:     0x6526fe7efdd7 - <unknown>
   7:     0x6526fe7ee9d9 - <unknown>
   8:     0x6526fe7efa64 - <unknown>
   9:     0x6526fe813393 - <unknown>
  10:     0x6526fe8137c6 - <unknown>
  11:     0x6526fe76ddd1 - <unknown>
  12:     0x6526fe76b793 - <unknown>
  13:     0x6526fe76b789 - <unknown>
  14:     0x6526fe7e6400 - <unknown>
  15:     0x6526fe76b7f5 - <unknown>
  16:     0x77e47162a1ca - <unknown>
  17:     0x77e47162a28b - __libc_start_main
  18:     0x6526fe76b77a - <unknown>
```

**Rerun version**
0.20.1

** Additional Context **
Currently we are bypassing the problem by setting the `RERUN_IS_PUBLISHING_CRATES` flag. But this can easily break in future versions :(

```
    "re_sdk": [crate.annotation(
        build_script_env = {
            "RERUN_IS_PUBLISHING_CRATES": "1",
        },
    )],
    "re_analytics": [crate.annotation(
        build_script_env = {
            "RERUN_IS_PUBLISHING_CRATES": "1",
        },
    )],
    "rerun": [crate.annotation(
        build_script_env = {
            "RERUN_IS_PUBLISHING_CRATES": "1",
        },
    )],
    "re_video": [crate.annotation(
        build_script_env = {
            "RERUN_IS_PUBLISHING_CRATES": "1",
        },
    )],
```

olivewong, 2024-11-27:
having this as well, also using bazel

teh-cmc, 2024-11-28:
Hijacking `RERUN_IS_PUBLISHING_CRATES` is actually generally the right idea, although as you said we've never committed to publicly stabilize that one.
We could make the situation nicer by having a dedicated check for Bazel though, similar to what we do for Conda builds.

Is there any environment variable you would recommend we check, in order to know whether we are part of a Bazel build?

idobenamram, 2024-11-28:
hmm, im not an expert in bazel's implementation of `rules_rust` but i'm not sure they add anything specific. what about not failing if `cargo metadata` is not found, is that something you think would be ok?

teh-cmc, 2024-11-28:
We can definitely start there yes if that makes users lives easier for now. Though we'll want something more streamlined at some point -- there's a lot more subtleties to get right beyond cargo-metadata when it comes to these ecosystem-specific builds.

teh-cmc, 2024-11-28:
Could you try and point your bazel build towards this PR @idobenamram please? That should do the trick, hopefully.

teh-cmc, 2024-11-28:
Ha yes, that's because `main` is in the middle of a big migration that involves monkey-patched crates. But that should mean things are working, or you probably wouldnt even get to that point.

I'll make a backported branch off of `0.20.1` to do one final test, and then we might be able to do a tiny patch release.

teh-cmc, 2024-11-28:
There is now a `release-0.20.2` branch, which is just `0.20.1` + #8239. Could you try that @idobenamram please? If that works out we can ship a patch.

idobenamram, 2024-11-29:
thanks again!