rerun-io/rerun #8233: Project: Performant visualization of scenes with large number of entities

teh-cmc, 2024-11-27:
## Context

We want the viewer to be able to scale to scenes with large numbers of entities.
This of course means visualizing these scenes, but also ingesting them in the first place.

This is blocked on a number of specific implementation issues, but put broadly: the work the viewer has to do to layout a scene more often than not grows linearly with the number of entities present in the _entire_ dataset.

There are only two ways to combat this:
* Break the linear growth where possible (e.g. only compute the data for the visible part of a list as opposed to the entire list).
* (Incremental) caching where possible (e.g. incrementally cache and update the transform hierarchy).

Of course in many cases, option 1 isn't even an option: if the user wants to visualize all entities in the scene, then somehow we have to make that fast.

Incremental caching of aggregated data (which is what the visualizers work with) is very hard, but will be a must in order to reach our performance goals. 

This issue is *not* about:
* Making the UX work for large number of entities (search, etc).

## Measurable end goals

### Air Traffic example (`2h` dataset)

* Ingestion directly from the SDK should be bottlenecked by the SDK, always.
* Ingestion from file should be order of magnitude faster, in the single digit seconds realm.
* Should be able to visualize, without visible time range, at or close to 60FPS on any decent machine, on the web.
* Should be able to visualize, with infinite visible time range, to run at 60FPS on high-end machines, on native.

### Revy

Revy was infamously bottlenecked by the performance of many entities (game scenes have _a lot_ of them).
This is a good opportunity to revive that project, if we can make it happen.

* Should be able to ingest and visualize Bevy's "Alien Cake Addict" example in real-time, on a decent machine, on the web.

## Relevant material

Writings:
* https://github.com/rerun-io/rerun/issues/5974
* https://github.com/rerun-io/rerun/issues/8221
* TODO(cmc): link to updated range semantics proposal
* TODO(cmc): link to range-zipped caching proposal

PRs:
* https://github.com/rerun-io/rerun/pull/5449
* https://github.com/rerun-io/rerun/pull/8224

## Sub-issues

_To be replaced with actual sub-issues once/if we have access to them. We can also remove the `project-many-entities-perf` label at that point._

* https://github.com/rerun-io/rerun/issues/4298
* https://github.com/rerun-io/rerun/issues/7025
* https://github.com/rerun-io/rerun/issues/7084
* https://github.com/rerun-io/rerun/issues/7604
* https://github.com/rerun-io/rerun/issues/7678
* https://github.com/rerun-io/rerun/issues/8101

teh-cmc, 2024-12-02:
What does performance look like when we only do the querying + chunk processing part? I.e. let's pretend we've managed to optimized out everything that visualizers do after receiving the chunks (range-zipping, color/radii/etc splatting, annotation context...).

We need to know this because our upcoming aggregated caching plan still involves running the queries every frame.

I've applied the following patch, which (very broadly) simulates that:
<details>
  <summary>Click for diff</summary>

```diff
diff --git a/crates/viewer/re_space_view/src/results_ext.rs b/crates/viewer/re_space_view/src/results_ext.rs
index 55e52be3e00..01d5cae4e17 100644
--- a/crates/viewer/re_space_view/src/results_ext.rs
+++ b/crates/viewer/re_space_view/src/results_ext.rs
@@ -427,7 +427,7 @@ impl<'a> HybridResultsChunkIter<'a> {
     pub fn component<C: re_types_core::Component>(
         &'a self,
     ) -> impl Iterator<Item = ((TimeInt, RowId), ChunkComponentIterItem<C>)> + 'a {
-        self.chunks.iter().flat_map(move |chunk| {
+        self.chunks.iter().filter(|_| false).flat_map(move |chunk| {
             itertools::izip!(
                 chunk.iter_component_indices(&self.timeline, &self.component_name),
                 chunk.iter_component::<C>(),
@@ -441,7 +441,7 @@ impl<'a> HybridResultsChunkIter<'a> {
     pub fn primitive<T: arrow2::types::NativeType>(
         &'a self,
     ) -> impl Iterator<Item = ((TimeInt, RowId), &'a [T])> + 'a {
-        self.chunks.iter().flat_map(move |chunk| {
+        self.chunks.iter().filter(|_| false).flat_map(move |chunk| {
             itertools::izip!(
                 chunk.iter_component_indices(&self.timeline, &self.component_name),
                 chunk.iter_primitive::<T>(&self.component_name)
@@ -458,7 +458,7 @@ impl<'a> HybridResultsChunkIter<'a> {
     where
         [T; N]: bytemuck::Pod,
     {
-        self.chunks.iter().flat_map(move |chunk| {
+        self.chunks.iter().filter(|_| false).flat_map(move |chunk| {
             itertools::izip!(
                 chunk.iter_component_indices(&self.timeline, &self.component_name),
                 chunk.iter_primitive_array::<N, T>(&self.component_name)
@@ -475,7 +475,7 @@ impl<'a> HybridResultsChunkIter<'a> {
     where
         [T; N]: bytemuck::Pod,
     {
-        self.chunks.iter().flat_map(move |chunk| {
+        self.chunks.iter().filter(|_| false).flat_map(move |chunk| {
             itertools::izip!(
                 chunk.iter_component_indices(&self.timeline, &self.component_name),
                 chunk.iter_primitive_array_list::<N, T>(&self.component_name)
@@ -489,7 +489,7 @@ impl<'a> HybridResultsChunkIter<'a> {
     pub fn string(
         &'a self,
     ) -> impl Iterator<Item = ((TimeInt, RowId), Vec<re_types_core::ArrowString>)> + 'a {
-        self.chunks.iter().flat_map(|chunk| {
+        self.chunks.iter().filter(|_| false).flat_map(|chunk| {
             itertools::izip!(
                 chunk.iter_component_indices(&self.timeline, &self.component_name),
                 chunk.iter_string(&self.component_name)
@@ -503,7 +503,7 @@ impl<'a> HybridResultsChunkIter<'a> {
     pub fn buffer<T: arrow::datatypes::ArrowNativeType + arrow2::types::NativeType>(
         &'a self,
     ) -> impl Iterator<Item = ((TimeInt, RowId), Vec<re_types_core::ArrowBuffer<T>>)> + 'a {
-        self.chunks.iter().flat_map(|chunk| {
+        self.chunks.iter().filter(|_| false).flat_map(|chunk| {
             itertools::izip!(
                 chunk.iter_component_indices(&self.timeline, &self.component_name),
                 chunk.iter_buffer(&self.component_name)
diff --git a/crates/viewer/re_space_view_spatial/src/visualizers/utilities/entity_iterator.rs b/crates/viewer/re_space_view_spatial/src/visualizers/utilities/entity_iterator.rs
index 0e535138677..3ba2cd3a26f 100644
--- a/crates/viewer/re_space_view_spatial/src/visualizers/utilities/entity_iterator.rs
[... 64 lines cut ...]

`main`, latest-at, with plot:
[image]

`main`, infinite range for 3D view / latest-at for the rest, without plot:
[image]

`main`, infinite range for 3D view / latest-at for the rest, with plot:
[image]

---

Now here's where it gets interesting: consider what happens 

`main`, latest-at, without plot -- **Chunk processing only**:
[image]
[image]
(NOTE: The specific values is the flamegraph are inflated due to probing overhead).

`main`, infinite range for 3D view / latest-at for the rest, without plot -- **Chunk processing only**:
[image]

---

It looks like we can definitely afford to run the queries every frame, as long as we manage to make aggregated caching work.

teh-cmc, 2025-01-15:
This is a very nice, user-provided real-world benchmark we can use to drive these optimizations forward (auth required): https://drive.google.com/file/d/1ESejr2yAhEjyKYFD2u-wQanckK8OwzrN/view?usp=sharing

rasmusgo, 2025-02-22:
I'm really struggling because of this issue:

[image]

Wumpf, 2025-02-22:
oh interesting, didn't realize that this one still bubbles up, thought we weeded it out!
Which Rerun version and can you tell more about your scene, I figure lots of entities with images?

rasmusgo, 2025-02-26:
I invited you and the rerun org to https://github.com/rasmusgo/puzzle_bot if you want to take a look. I also uploaded a recording to [google drive](https://drive.google.com/file/d/1qMcs6ZkIWwD7cHjr9t8H0ya1PSTj1Ejl/view?usp=sharing).

Wumpf, 2025-02-26:
awesome, thank you! This week is a bit packed, I'll try to have a browse next week, surely this must be something silly in the viewer that can't be hard to fix 😄 🤞

rasmusgo, 2025-03-17:
Any progress on this?

Wumpf, 2025-03-17:
@rasmusgo sorry for the long delay and thanks for the ping (had this on my radar but kept postponing :/)!
There was indeed a lot of sillyness, addressed [here](https://github.com/rerun-io/rerun/pull/9308)