rerun-io/rerun #8245: Use a software renderer for kittest

abey79, 2024-11-28:
Kittest-based UI snapshot test currently use whatever hardware backend wgpu picks, which has multiple drawbacks:
- each backend may produce a slightly different output, making comparison more difficult
- a hardware backed is required, which rules out windows and linux GHA runners

Using a software rendering backend would solve both of these issues.

emilk, 2024-11-29:
> each backend may produce a slightly different output, making comparison more difficult

According to @Wumpf this is likely due to differences in filtering. If we use a manual four-tap bilinear filter in the egui pixel shader we should get a lot closer results, hopefully within the error margins.

Wumpf, 2025-01-07:
> Using a software rendering backend would solve both of these issues.

Unfortunately not true since software renderer among each other also have different results. E.g. llvmpipe won't give use the same as microsoft's WARP

emilk, 2025-01-07:
So we're back to the manual-texture-filtering idea then?

Wumpf, 2025-01-07:
That would surely alleviate the issue(s), but we'll for sure never catch all differences anyways 🤷. I don't think there's a remedy for tweaking error thresholds

PR #8620 (2025-01-10): Run Linux & Windows screenshot tests via `lavapipe` software rasterizer
### Related

* Fixes #8245 

### What

Enables screenshot tests on Linux & Windows ci using Mesa lavapipe (the Vulkan version of llvmpipe) software rasterizer.
Uses wgpu's Mesa build https://github.com/gfx-rs/ci-build (thanks @cwfitzgerald!)

ICD driver file discovery & how to best download the Vulkan SDK was giving me a bit of a headache (via apt as done on wgpu's ci was very slow for some reason and wgpu only does GL software rasterizing on Windows). But it's all figured out now & documented in the setup script :). The p [... cut]
closed 2025-01-10 by Wumpf