rerun-io/rerun #8247: Docs.rs for 0.20.2 are missing (time out)

teh-cmc, 2024-11-29:
See build logs here: https://docs.rs/crate/rerun/0.20.2/builds/1567546.

The worker times out after 30 minutes:
```
[INFO] [stderr]  Documenting rerun v0.20.2 (/opt/rustwide/workdir)
[ERROR] error running command: command timed out after 1800 seconds
```

teh-cmc, 2024-11-29:
I wish we could just compute the docs on our end and upload the final artifact to docs.rs... :thinking:

emilk, 2024-11-29:
Oh God, how is 30 minutes not enough?
* https://github.com/rust-lang/docs.rs/issues/2638

emilk, 2024-11-29:
* Yet another reason to work on https://github.com/podusowski/walkers/issues/195

emilk, 2024-12-01:
Timeout has been increased, and the docs are up: https://docs.rs/crate/rerun/0.20.2