rerun-io/rerun #8286: Support arbitrary X axis in the plot view using a `ScalarAbscissa`

abey79, 2024-12-03:
This [discord thread](https://discord.com/channels/1062300748202921994/1312444177304256512/1312444177304256512) got me thinking and I'd like to suggest this alternative take on:
- https://github.com/rerun-io/rerun/issues/5622

The core proposal is to introduce a `ScalarAbscissa` component (someone please finds a better name) that is essentially a `Scalar` component to be used as X axis (with [tagged components](#6889) we could actually reuse the same `Scalar` component).

Then, the existing Timeseries view would be altered as follows:

- Renamed to `Plot view`.
- When used in Latest At mode: plot the `Scalar` component (which, in this case, should typically be a n>1 batch) against the `ScalarAbscissa` component. 
- When used in Range mode: no modifications w.r.t to the current behaviour. (For a given time point, if the `Scalar` is a batch, pick instance at index 0 and ignore others. Also, ignore `ScalarAbscissa`.)

Bonus:
- When in Range mode with n>1 `Scalar` batches, have some options to display a compound value for each point (mean/media with stddev bars, box plot, etc.)

gavrelina, 2024-12-03:
related to https://github.com/rerun-io/rerun/issues/8286

grtlr, 2024-12-05:
Very interesting! In general, I would envision Reruns plots + blueprints to evolve into something that resembles [Vega-Lite](https://vega.github.io/vega-lite/) which should give users a lot of flexibility and control over how their data is visualized. 

Vega-Lite is a declarative grammar (that should map very well to our concept of blueprints) for creating visualizations.  

Here is [an example](https://vega.github.io/vega-lite/docs/line.html#connected-scatter-plot) of how plotting a trajectory looks like in VegaLite:

```json
{
  "data": {"url": "data/driving.json"},
  "mark": {"type": "line", "point": true},
  "encoding": {
    "x": {
      "field": "miles", "type": "quantitative",
      "scale": {"zero": false}
    },
    "y": {
      "field": "gas", "type": "quantitative",
      "scale": {"zero": false}
    },
    "order": {"field": "year"}
  }
}
```

Above code yields the following plot:

[image]

abey79, 2024-12-05:
It looks like entity reference would be needed to fully support this model:
- https://github.com/rerun-io/rerun/issues/2682

benliepert, 2025-05-15:
> the use case where the entire plot (X and Y axis) is stored at a single time point

+1 For this.

Specifically, we have an FPGA monitoring an RF envelope that produces a vector per higher level timestep.
It would be nice to visualize this vector as a 2D plot per time point in a way that preserves all the data.

Hardliu, 2026-01-04:
Has this function been realized? @abey79

abey79, 2026-01-04:
> Has this function been realized?

Not yet unfortunately.