rerun-io/rerun #8293: Tagged components milestone 3: descriptor-driven and partial queries

teh-cmc, 2024-12-03:
The third milestone for tagged components is to be able to query (`LatestAt` and `Range`) for components based on descriptor tags and potentially partial information.
The semantics of the viewer must not be impacted in any way by that change.

E.g.:
* `LatestAt("a/b/c@*:Color#*)`, which is what you can do today.
* `LatestAt("a/b/c@LineStrips2D:Color#*)`, to ask specifically for colors belonging to `LineStrips2D`.
* `LatestAt("a/b/c@LineStrips2D:Color#edge_color)`, to ask specifically for the edge colors.

This is the last milestone before people can actually start building features on top of tagged components.

More details TBD.

---

* Part of https://github.com/rerun-io/rerun/issues/6889
* Requires #7948

grtlr, 2025-05-23:
I think we should change these queries to be based on `archetype_field_name` similar to #9906.

grtlr, 2025-05-23:
I've created an issue to track this:

* [OSS-1285](https://linear.app/rerun/issue/OSS-1285/change-dataframe-queries-to-use-archetype-field-name-instead-of)

closed 2025-05-23 by grtlr