rerun-io/rerun #8338: Improve our copy to clipboard story

abey79, 2024-12-06:
It's relatively easy to copy stuff from the UI (pasting is another story), and can occasionally be very useful. 

There are at least two classes of interactions:
- Copy "whatever is selected" upon cmd-C. It can get weird with multi-selection, but for single selection, especially of entities, it would be very useful to simply copy the entity path.
- Add "copy" button in strategic places.
  - Next to time stamp display, to copy a full timestamp with standard-compliant formatting
    [image]
  - ...other places?

Related:
- https://github.com/rerun-io/rerun/issues/6015

emilk, 2024-12-06:
When a space view is selected, copying could copy both a descriptive string, and a screenshot :)
* Related: https://github.com/emilk/egui/issues/5424

nikolausWest, 2024-12-06:
Another perhaps simpler, perhaps not, thing is just making more data selectable as text so you can copy the text.

teh-cmc, 2024-12-06:
At least in the selection and time panels, I've often wanted clicking on a (partial) entity path to copy the full entity path to clipboard.