rerun-io/rerun #8344: entity_path_prefix not working for log_file()

EtaLoop, 2024-12-06:
**Describe the bug**
`entity_path_prefix` does not seem to work with `rerun.log_file_from_path` on 0.20.3.

**To Reproduce**
```
rec = rr.new_recording("app", spawn=True)

rr.log_file_from_path(
    "cube.stl",
    entity_path_prefix="test",
    recording=rec.to_native(),
)

```

teh-cmc, 2024-12-09:
I don't believe we have a builtin STL loader, so I assume this is a custom one you made.

All `entity_path_prefix` does is pass a flag of the same name to your dataloader -- from there it's your job to do whatever you'd like with that information.

See https://rerun.io/docs/reference/data-loaders/overview:
[image]

(Closing, feel free to re-open if needed)

EtaLoop, 2024-12-09:
Here : https://rerun.io/docs/getting-started/data-in/open-any-file it says that stl is supported.
So I am wrong if I'm saying that there is a DataLoader for stl ?