rerun-io/rerun #8345: Documentation mentioning `pip3 install` might not work on windows and Mac as `pip3` might reference user pip instead of environment pip

haixuanTao, 2024-12-06:
Just want to point out that some of the doc mentioning:

```bash
    Failed to find Rerun Viewer executable in PATH.

        You can install an appropriate version of the Rerun Viewer via binary releases:
        * Using `cargo`: `cargo binstall --force rerun-cli@0.19.1` (see https://github.com/cargo-bins/cargo-binstall)
        * Via direct download from our release assets: https://github.com/rerun-io/rerun/releases/0.19.1/
        * Using `pip`: `pip3 install rerun-sdk==0.19.1`
   ```

Might not fix the problem of path as pip3 is going to be usr/bin/pip3 instead of an environment pip such as `/Users/xaviertao/miniconda3/bin/pip`. For more info check out: https://discussions.apple.com/thread/255428578?sortBy=rank

I would recommend either have a OS specific error message and make sure that readme in the rerun documentation mention this possibility.

Had this problem in dora,

Wumpf, 2024-12-09:
Whether to use pip or pip3 depends a lot on your environment, not just the OS. The reason we write pip3 there is exactly because on MacOS that's the option that works out of the box. The assumption is that if you already use a custom environment you know what you're dealing with.
That is, there's really no reliable way to detect whether pip3 or pip is the correct thing to do.

haixuanTao, 2024-12-09:
I understand your point but just want to point out that pip3 does not work out of the box on macOS and Windows, quite literally, as the default path from pip3 will not resolve, where as pip will. 

And it might be a hard one to debug for someone beginning. 

But I can understand that you want to ignore this.

Wumpf, 2024-12-09:
> I understand your point but just want to point out that pip3 does not work out of the box on macOS and Windows

On macos the default installed python is only reachable via `pip3` nowadays last time I checked, maybe that changed again? So if you don't install any python environment yourself (in your case you're apparently using miniconda), then `pip3` seems to be the right one to pick as I understand.
On windows either will bring up the "install from store" suggestion

haixuanTao, 2024-12-09:
Yes, so the thing is that the default `pip3` indeed exist but installing scripted binaries will not be added to PATH by default.
https://github.com/user-attachments/assets/a12ebb75-c1e5-46f4-b5a3-92819c102214

Here is an example with `black` binary. 

And this is also going to fell if you're within a virtual environment.

But I understand that putting pip will not work by default as well as it might be installed properly.

Wumpf, 2024-12-10:
Thanks for digging into this, that's a compelling argument didn't quite grasp this earlier! Sounds like the default installed pip/python is no use here anyways 😬

dubing-bit, 2025-07-28:
I have the same problem in ubuntu22.04
```
RuntimeError: Failed to find Rerun Viewer executable in PATH.

    You can install an appropriate version of the Rerun Viewer via binary releases:
    * Using `cargo`: `cargo binstall --force rerun-cli@0.24.0` (see https://github.com/cargo-bins/cargo-binstall)
    * Via direct download from our release assets: https://github.com/rerun-io/rerun/releases/0.24.0/
    * Using `pip`: `pip3 install rerun-sdk==0.24.0`

    For more information, refer to our complete install documentation over at:
    https://rerun.io/docs/getting-started/installing-viewer
```