rerun-io/rerun #8377: Error with webviewer

praveen-aurora, 2024-12-09:
**Describe the bug**
Got error with web viewer:
LinkError: WebAssembly.instantiate(): Import #10 "wbg" "__wbg_structuredClone_046ae3829d60ba4b": function import requires a callable

**To Reproduce**
Built the web viewer with 0.20.3 and tried to run it on of my .rrd files.

**Rerun version**
0.20.3

Wumpf, 2024-12-10:
@praveen-aurora how did you run & build the web viewer exactly?

praveen-aurora, 2024-12-10:
I checked out the repo at the 0.20.3 commit, and ran 

pixi run rerun-build-web-release

from the web_viewer subfolder. Then I uploaded the files in that folder to our internal web server and invoked the viewer that way. 

I also tried copying the web viewer assets from https://app.rerun.io/version/0.20.3/ to our internal web server and invoking the viewer, and got a similar error.

Wumpf, 2024-12-10:
Thanks! I just tried `git checkout 0.20.3 && pixi run rerun-web-release` (that's intentionally without the build, so I can test it in-place) on a Windows machine and it worked fine, so not sure what's up there 🤔
Afaik this error usually means that the accompanying js (the generated one at `./web_viewer/re_viewer.js`) isn't looking for the right things in the wasm blob. Did you make sure to update that on your web server as well? Otherwise (long shot) you might need to try doing a clean rebuild if you haven't already.

praveen-aurora, 2024-12-10:
Ok, I think the nuscenes one specifically isn't working. So I'm testing with the arkit one, which is.

praveen-aurora, 2024-12-10:
Ok, I tried out launching the viewer with the pixi command you used - that works for me to load the arkit example, but the nuscenes example and my own rrd both briefly show the UI and then immediately go black. So I think the nuscenes example is one that provides a good repro of the issue.

Wumpf, 2024-12-10:
Huh strange! app.rerun.io/version/0.20.3/index.html?url=https://app.rerun.io/version/0.20.3/examples/nuscenes_dataset.rrd works just fine for me. Tested it on Windows in Firefox & Chrome. Maybe it's a browser cache issue?

praveen-aurora, 2024-12-10:
Interestingly, that link works for me in Safari on Mac but not on Chrome on
Mac. I'll see about the cache in Chrome.

praveen-aurora, 2024-12-10:
The good news is that my rrd also works in Safari. However, even with clearing the cache in Chrome it doesn't work in Chrome.

sashankaryal, 2024-12-10:
I'm having the same issue on Mac. The `nuscenes` dataset works in Firefox 133.0b2 (aarch64) but not in chrome version 131.0.6778.140. I tried it in private window, too, but no luck. I've screenshotted and attached my browser's logs:

[image]

Wumpf, 2024-12-11:
@sashankaryal that's a different but also very concerning issue: it looks like chrome's webgpu impl no longer accepts one of our shaders. To narrow this down a bit, what scene did you run there?

Wumpf, 2024-12-11:
If you force the webgl renderer like so https://rerun.io/viewer?url=https%3A%2F%2Fapp.rerun.io%2Fversion%2F0.20.3%2Fexamples%2Fnuscenes_dataset.rrd&renderer=webgl I reckon it still works?

sashankaryal, 2024-12-11:
Yes, it works if I force the webgl renderer. I restarted my browser, downgraded, upgraded back again, tried in private mode, no help. What's confusing is that it works on other recordings, like ARKit, although it does complain about some wgpu error. I've attached a screenshot. With nuscenes, it doesn't even get to that point.

[image]

Wumpf, 2024-12-11:
Thanks for going through the trouble and trying all these. Seems to be a very similar error though. In either case Chrome has trouble with one of our wgsl shaders 🤔.  Very strange indeed! The error itself sounds a bit like its happening once Chrome is finished translating to MSL (metal shading language) and then wants to compile it, so maybe it's OS version dependent? My M1 is still on 14.4.1

yakunouyang, 2025-02-27:
Hi, I encounter the same problem with [this example](https://rerun.io/viewer?url=https://app.rerun.io/version/0.22.1/examples/rrt_star.rrd) (Chrome 133.0.6943.141, macOS version: 15.2), run the chrome browser with dawn flag `--enable-dawn-features=dump_shaders,disable_symbol_renaming`, resulting in following error which is more clear:

[image]

https://github.com/rerun-io/rerun/blob/af3312a21fb157b76dce2c509a1934ad252d1b67/crates/viewer/re_renderer/shader/point_cloud.wgsl#L134

it seems that the variable distance collide with internal function distance(), I've created a PR to resolve this #9152 , hope it helps.

yakunouyang, 2025-02-27:
> re_viewer.js:3320 ERROR: [re_renderer::error_handling::error_tracker] re_renderer/src/error_handling/error_tracker.rs:158: Wgpu validation error 2: Unable to create library object: program_source:165:46: error: called object type 'float' is not a function or function pointer float const tint_symbol_85 = tint_symbol_85(tint_symbol_87.tint_symbol_56, tint_symbol_87.tint_symbol_57, tint_symbol_87.tint_symbol_75, tint_module_vars); ~~~~~~~~~~~~~~^ .
> 
> * While calling [Device "re_renderer device"].CreateRenderPipeline([RenderPipelineDescriptor]).

[image]

reproduced in Chrome 131, with `--enable-dawn-features=dump_shaders,disable_symbol_renaming` shows that it's another variable naming bug, I think with another re-naming fix this issue can be fully closed? @Wumpf 

[image]

after fix:

[image]

Wumpf, 2025-03-03:
@yakunouyang I think you're right - I didn't realize that the original issue was this very shader compilation issue in Chrome 131 as it was buried under all that log

PR #9160 (2025-03-03): Fix shader compilation further in Chrome 131
### Related

* Closes #8377
* Follow-up to  #9152

### What

This pr is extension of #9152, further fixing the shader variable naming issue in Chome 131 version.

[image]

after fix:

[image]
commit 81e1e0fc (2025-03-03): Fix shader compilation further in Chrome 131 (#9160)

### Related

* Closes #8377
* Follow-up to  #9152

### What

This pr is extension of #9152, further fixing the shader variable naming
issue in Chome 131 version.

<img width="1021" alt="Image"
src="https://github.com/user-attachments/assets/fab624cd-3d14-4bbc-8209-5a1cc1ccbd73"
/>

after fix:

<img width="1469" alt="Image"
src="https://github.com/user-attachments/assets/3f08da34-919d-4efb-9104-137dd6fe2863"
/>
closed 2025-03-03 by Wumpf