rerun-io/rerun #8389: Debug-panic when cloning space views

emilk, 2024-12-10:
Open the following .rbl file with rerun: [bug.rbl.zip](https://github.com/user-attachments/files/18080529/bug.rbl.zip)

> thread 'main' panicked at 'Insert Chunk with multiple values for component named `rerun.blueprint.components.QueryExpression`: this is currently UB'
re_chunk_store/src/writes.rs:42

The .rbl was created like this:
[image]

teh-cmc, 2024-12-10:
The issue is this chunk, which indeed is doing weird shenanigans:
```
Chunk(180FD546F6D61F7827EC6937C19E6022) with 2 rows (1.1 KiB) - /space_view/06969201-2f22-43a0-a598-0c737d0b21e3/SpaceViewContents - 
┌─────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┐
│ CHUNK METADATA:                                                                                                                             │
│ * entity_path: "/space_view/06969201-2f22-43a0-a598-0c737d0b21e3/SpaceViewContents"                                                         │
│ * heap_size_bytes: "968"                                                                                                                    │
│ * id: "180FD546F6D61F7827EC6937C19E6022"                                                                                                    │
│ * is_sorted: ""                                                                                                                             │
├╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌┤
│ ┌──────────────────────────────────┬───────────────┬───────────────────┬─────────────────────────────────────┬────────────────────────────┐ │
│ │ RowId                            ┆ blueprint     ┆ QueryExpression   ┆ QueryExpression                     ┆ SpaceViewContentsIndicator │ │
│ │ ---                              ┆ ---           ┆ ---               ┆ ---                                 ┆ ---                        │ │
│ │ type: "TUID"                     ┆ type: "i64"   ┆ type: "list[str]" ┆ type: "list[str]"                   ┆ type: "list[null]"         │ │
│ │ kind: "control"                  ┆ is_sorted: "" ┆ kind: "data"      ┆ archetype_field_name: "query"       ┆ kind: "data"               │ │
│ │                                  ┆ kind: "time"  ┆                   ┆ archetype_name: "SpaceViewContents" ┆                            │ │
│ │                                  ┆               ┆                   ┆ kind: "data"                        ┆                            │ │
│ ╞══════════════════════════════════╪═══════════════╪═══════════════════╪═════════════════════════════════════╪════════════════════════════╡ │
│ │ 180FD546F6C917DB27EC6937C19E6011 ┆ 4             ┆ [+ $origin]       ┆ -                                   ┆ [-]                        │ │
│ ├╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌┼╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌┼╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌┼╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌┼╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌╌┤ │
│ │ 180FD546F6D2706F27EC6937C19E601B ┆ 4             ┆ -                 ┆ [+ $origin]                         ┆ [-]                        │ │
│ └──────────────────────────────────┴───────────────┴───────────────────┴─────────────────────────────────────┴────────────────────────────┘ │
└─────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┘
```

That's probably because we're fetching an old blueprint from somewhere, and then modifying it at runtime.

teh-cmc, 2024-12-10:
Yeah no point in overengineering this, I'll probably just remove the assertion.

Unless maybe I can ignore it only for blueprints for now.

closed 2024-12-10 by teh-cmc

teh-cmc, 2024-12-11:
Actually, now that I think about it: we cannot just ship with UB if users load old blueprints 😅 This will inevitably lead to some nightmarish, irreproducible, subtle bugs.

We need to not only ignore the assertion, but also get rid of the old untagged data when this happens.

PR #8393 (2024-12-10): Store: only run the debug assertion against recordings (not blueprints)
I'd like to keep that assertion a bit longer, just in case. So for now just ignore blueprints, as they are problematic in that case.

* Fixes #8389
commit a826845f (2024-12-10): Store: only run the debug assertion against recordings (not blueprints) (#8393)

I'd like to keep that assertion a bit longer, just in case. So for now
just ignore blueprints, as they are problematic in that case.

* Fixes #8389