rerun-io/rerun #8402: Isolate individual time series data by hiding all others

jaraujo98, 2024-12-11:
For my current project I need to stream robot reference and joint data. The time series view serves my purpose perfectly, but if I want to isolate a single joint I need to manually hide all others. Having the option to double click (or right click) the visibility icon to isolate that entity would be very helpful. I'm happy to give a shot at this, if the maintainers agree it would be a useful feature.

teh-cmc, 2024-12-11:
Quickly finding and isolating / focusing on specific entities in the viewer is something we're actively working towards! No date yet, but we'll make sure to update this issue.

cc @gavrelina @abey79

abey79, 2024-12-11:
We definitely want to improve things on this front, but it's already possible to hide all-but-one series from a timeseries view by alt-clicking in the legend:

https://github.com/user-attachments/assets/60a00395-c834-438a-8723-a149d2719acd

We should probably advertise this feature more.

abey79, 2024-12-11:
https://github.com/rerun-io/rerun/pull/8406