rerun-io/rerun #8411: Choose a good `max_decoding_message_size` for gRPC

jprochazk, 2024-12-11:
The default `StorageNodeClient` decoding size is 4 MB, which is too small as some of our messages can be orders of magnitude larger than that. We need to choose a better value, as it's currently set to `usize::MAX` to work around that

- https://github.com/rerun-io/rerun/pull/8347#discussion_r1879868667

emilk, 2025-02-14:
Our arrow buffers are currently limited in size by `i32`, so something like `u32::MAX` would probably make sense. If we make it too large, than we can accidentally try to allocate exabytes of memory and get hard crashes.

grtlr, 2025-02-14:
😬 https://github.com/rerun-io/rerun/blob/main/crates/store/re_grpc_client/src/redap/mod.rs#L116-L117

PR #9083 (2025-02-19): Move URI handling logic to new `re_uri` crate
### Related

* Cleanup of #9064
* Adds `/proxy` logic from #8876
* Closes #8411

### What

Moves the `address.rs` logic out of `re_grpc_client` into its own crate (because you can never have enough crates 📦).

Additionally, we try to follow the mantra of _parse, don't validate_ here and pass around `CatalogEndpoint` and `RecordingEndpoint` objects, rather than URI `String`s. Doing so gives us stronger type guarantees and avoids having to re-validate in some cases.

### Limitations

_As of now, opening catalog endpoints via the web viewer is not sup [... cut]