rerun-io/rerun #8412: Try using Arrow IPC compression

jprochazk, 2024-12-11:
We're currently doing our own LZ4 block compression for messages. Maybe it'd be better to use the one from Arrow IPC? The Arrow IPC `StreamWriter` has a `compression` option which supports LZ4.

- https://github.com/rerun-io/rerun/pull/8347#discussion_r1879874587

teh-cmc, 2024-12-11:
See https://rerunio.slack.com/archives/C04DLS7CCE4/p1732715341625009

zehiko, 2024-12-11:
> See https://rerunio.slack.com/archives/C04DLS7CCE4/p1732715341625009

thanks @teh-cmc, I forgot that we already discussed this.

jprochazk, 2024-12-11:
The reasons from that Slack thread:
- We want the compression to be separate from the encoding, because we may want something other than Arrow IPC
- We don't what Arrow IPC's compression gives us (such as per-column compression settings)
- We want to be able to change the compression scheme on-the-fly

In short, we gain very little and lose control, so it's not worth it.

closed 2024-12-11 by jprochazk