rerun-io/rerun #8417: rr.serve_web() deadlocks when logging before calling rr.serve()

jleibs, 2024-12-11:
Simple repro:
```python
 import rerun as rr
import time

rr.init("stuck")

rr.log("entity", rr.Points3D([[0, 0, 0], [1, 1, 1]], radii=[0.5, 0.5, 0.5]))

rr.serve()

print("Exiting") # Never output
```

By comparison:
```python
 import rerun as rr
import time

rr.init("stuck")

rr.serve()

rr.log("entity", rr.Points3D([[0, 0, 0], [1, 1, 1]], radii=[0.5, 0.5, 0.5]))

print("Exiting")
```
exits as expected.

jleibs, 2024-12-11:
This appears to have been an issue since 0.17.

MostafaOkasha, 2025-02-04:
Are there any workarounds to resolving this? Even killing my terminal and running my program again still hangs.

teh-cmc, 2025-02-04:
> Are there any workarounds to resolving this? Even killing my terminal and running my program again still hangs.

According to the PR description, the workaround is to call `serve_web` first (i.e. just after `init`). Does that not work for you?

MostafaOkasha, 2025-02-04:
Since I'm trying to manage multiple recording streams, I'm using `new_recording()` (in dicts) to track multiple streams.
Essentially, my call order is:

```python
recording_stream = rr.new_recording(...)
recording_stream.send_bluprint(blueprint, make_active=True, make_default=True)
recording_stream.serve(open_browser=False, ws_port=self.ws_port, server_memory_limit="4GB")
# Eventually...
recording_stream.log(...)
```

It wasn't always the case, but recently it's been hanging at the log `re_sdk::web_viewer] Hosting a web-viewer at http://localhost:9090?url=ws://localhost:9877`.

My regular workflow is that I am always killing and rerunning my program, and every 3/4 attempts, my terminal would get stuck at the log above. I would have to keep killing and trying again then it eventually works. Apologies if this is a separate problem, but I thought it might be similar.

teh-cmc, 2025-02-05:
Ha -- interesting.

Does swapping `send_blueprint` and `serve` help? I.e.
```py
recording_stream = rr.new_recording(...)
recording_stream.serve(open_browser=False, ws_port=self.ws_port, server_memory_limit="4GB")
recording_stream.send_bluprint(blueprint, make_active=True, make_default=True)
# Eventually...
recording_stream.log(...)
```