rerun-io/rerun #8530: `EntityPathFilter` has undefined behavior for sorting

Wumpf, 2024-12-18:
it's `Ord` and `Eq` don't check the same fields.
We heavily depend on this behavior in a bunch of places: we want equality to only depend on raw expressions but sorting should be done on expressions with variable substition (we sort from least specific to most specifiic - we want this for both evaluation and UI display)

-> We need to split those concerns into two different types, one using a hashmap of unresolved rules and the other one using a btreemap of resolved rules.

This can cause crashes when building with Rust 1.81+, see https://blog.rust-lang.org/2024/09/05/Rust-1.81.0.html#new-sort-implementations

Wumpf, 2024-12-18:
Users using the Rust dataframe api will always use unresolved entity path filter (this will be just called `EntityPathFilter`) whereas the viewer internally oftentimes want to use `ResolvedEntityPathFilter` which can only be created from a `EntityPathFilter` + substituions

rasmusgo, 2025-01-02:
Is this the issue that gives these crashes? I get these a lot when trying to move tabs around and sometimes just by hovering stuff in the blueprint pane.

```
thread 'main' panicked at 'user-provided comparison function does not correctly implement a total order'
core/src/slice/sort/shared/smallsort.rs:860
```

Wumpf, 2025-01-03:
@rasmusgo yes, that's exactly that. While testing the release we hit it only once and couldn't repro it again which is why we decided to move on :/. Also, it doesn't affect our python builds since they build with an older rust version that doesn't have that check yet.
Unfortunately, I wasn't able to fix this without breaking changes in the crate boundaries so far, so it can't go into a patch release

closed 2025-01-07 by Wumpf