rerun-io/rerun #8538: Ability to show edges grid on 3D mesh

dimentary, 2024-12-19:
**Is your feature request related to a problem? Please describe.**
The ability to show edges grid on top of the mesh, currently you could only show vertices on top of the mesh with the "Points3D" visualiser, which is not very informative (also you can't change points color separately from the mesh color):
[image]

**Describe the solution you'd like**
A simple toggle that once turned on shows edges on top of the mesh surface, the result should look something like this:
[image]

Wumpf, 2024-12-19:
> also you can't change points color separately from the mesh color

this will be soon addressed by tagged components, see also https://github.com/rerun-io/rerun/issues/6889

---

some notes on implementation:

**Datastructures:**
tbd if this should be a separate visualizer/archetype or just be a display option on `Asset3D` & `Mesh3D`. Leaning towards the later for ease of use (also the intricacies of "archetype inheritance", see https://github.com/rerun-io/rerun/issues/8368 might be be out too long).

**Rendering:**
Implementation-wise we won't be able to just simply draw the triangle mesh as wireframe since that's unfortunately not fully portable (see https://wgpu.rs/doc/wgpu/struct.Features.html#associatedconstant.POLYGON_MODE_LINE - this is a native-only wgpu feature). Using our line renderer is an option, but has too much overhead, we'd really like to use the data we already have.
Drawing from vertex+index buffer using vertex pulling (doing random buffer access in vertex & index buffer) is actually quite doable and that would allow us to render the lines _on_ the solid mesh, avoiding issues with lines intersecting the solid mesh when both solid & wire is drawn.
https://webgpu.github.io/webgpu-samples/?sample=wireframe is a nice implementation of just that and allows to switch between two variants, one with separate line geometry (note how this requires a depth bias which can be fiddly to configure) and one using barycentric coordinates.
**But** we can't do that on WebGL - let's check for alternatives & then decide whether it's worth feature-gating this.

julianschoep, 2024-12-20:
As a temporary work-around you can visualize the mesh connectivity by plotting line strips like so:

```python
rr.log(
    "mesh_grid",
    rr.LineStrips3D(
        mesh.vertices[mesh.faces],
        colors=[0, 0, 0],
        radii=0.0005,
    ),
    timeless=True,
)

```

Nikita240, 2025-03-06:
I made a workaround for wireframe rendering by using a custom texture and mapping the vertices to the edges of that texture.

The caveats are:
1. The wireframe thickness scales with the size of the triangle. This is only viable when you have a relatively constant triangle size.
2. It requires duplicate vertexes. No sharing vertices between triangles. 

```rust
fn with_wireframe_texture(mesh: rr::Mesh3D) -> rr::Mesh3D {
    // make sure incoming mesh has unique vertices for each triangle

    let (format, blob) = wireframe_texture();

    let mut vertex_texcoords = Vec::with_capacity(mesh.vertex_positions.len());

    for _tri in mesh
        .triangle_indices
        .as_ref().expect("to have tris").iter() {
            vertex_texcoords.push([1.0, 1.0]);
            vertex_texcoords.push([0.0, 1.0]);
            vertex_texcoords.push([0.0, 0.0]);
    }

    mesh.with_vertex_texcoords(vertex_texcoords)
        .with_albedo_texture(format, blob)
}

fn wireframe_texture() -> (rerun::ImageFormat, std::vec::Vec<u8>) {
    const SIZE: u32 = 100;
    // Fill the image with white.
    let mut texture_img = RgbImage::from_pixel(SIZE, SIZE, Rgb([255, 255, 255]));

    // Draw 1px black lines on edges.
    for i in 0..SIZE {
        texture_img.put_pixel(0, i, Rgb([0, 0, 0])); // Black line
        texture_img.put_pixel(i, 99, Rgb([0, 0, 0])); // Black line
        texture_img.put_pixel(i, i, Rgb([0, 0, 0])); // Black line
    }

    // My shitty attempt at antialiasing.
    for i in 0..SIZE - 1 {
        texture_img.put_pixel(i + 1, i, Rgb([255 / 5, 255 / 5, 255 / 5])); // antialiasing
        texture_img.put_pixel(i, i + 1, Rgb([255 / 5, 255 / 5, 255 / 5])); // antialiasing
    }

    (ImageFormat::rgb8([SIZE, SIZE]), texture_img.into_raw())
}
```

I wrote this for rerun 0.18. Syntax might be a bit different for latest.

[image]