rerun-io/rerun #8559: Investigate `re_renderer` failure with Safari opt-in WebGPU support

abey79, 2024-12-23:
On Sequoia, WebGPU is now opt-in with a feature flag (Develop menu -> Features Flags… -> WebGPU). However, most examples fail to properly render and Safari reports various validation errors:

[image]

Might be worth investigating in preparation for proper WebGPU hopefully coming soon.

Wumpf, 2025-05-15:
We have exactly one callsite for `copyBufferToTexture`: https://github.com/rerun-io/rerun/blob/a419c2eb2c56d6840fe8a86f86977df2beaa57e9/crates/viewer/re_renderer/src/allocator/cpu_write_gpu_read_belt.rs#L302

Note that at this time, the buffer is indeed still mapped! However, we ensure that all buffers in the write belt are unmapped **before** enqueuing the encoder. So it _could_ (!!) be that WebKit/Safari just validates too early!

At this point we should really just file a bug with this information. Providing a repro would be great, but might take a bit to write.

Wumpf, 2025-05-15:
Filed a bug against WebKit https://bugs.webkit.org/show_bug.cgi?id=293062

Wumpf, 2025-05-19:
@mwyrzykowski is being awesome and has a fix in the pipeline https://github.com/WebKit/WebKit/pull/45459

jprochazk, 2025-05-23:
Since this is an external issue and thus not actionable on our side, I'm removing it from the milestone.

Wumpf, 2025-07-12:
Confirmed we're now working with WebGPU in the Safari Technology Preview 223 (WebKit 20622.1.18.11.5)

Wumpf, 2025-07-12:
Created a follow-up ticket for (re)enabling WebGPU on Safari
https://github.com/rerun-io/rerun/issues/10609

closed 2025-07-12 by Wumpf