rerun-io/rerun #8560: Python vs Rust Transform3D does not match

alik-git, 2024-12-24:
**Describe the bug**  

What I'm doing: loading a urdf file in rerun in order to visualize a humanoid robot. 

Issue summary: The code works in python, but as far as I can tell I'm doing the exact same thing in rust (exact same log calls to rerun in the same order) but it does not match. The joint rotations are incorrect when you use the rust code, see pictures below. Furthermore, when I save the visualization from python as an RRD, and then inspect it  manually (dataframe -> csv) it doesn't seem to match the actual rerun log calls my script made, which is especially confusing since with python the robot looks correct.

So I'd just like to know if I'm doing something wrong myself, or the rust api and python api are different, or reading the rrd file via the dataframe api is not quite right, or something else, but this is a subtle issue so I'm making it on github instead of the discord.

[image]

[image]

[image]

[image]

[image]

[image]

**To Reproduce**  
1. Clone these two repos/branches and follow the readme install instructions:
   - [Rust code (debug_transform branch)](https://github.com/alik-git/krecviz_rust/tree/debug_transform)
   - [Python code (debug_rust_transform branch)](https://github.com/kscalelabs/krecviz/tree/debug_rust_transform)
2. Run the Python code:
   ```bash
   python visualize.py --urdf data/urdf_examples/gpr/robot.urdf --krec data/krec_examples/actuator_22_right_arm_shoulder_roll_movement.krec
   ```
3. Run the Rust code:
   ```bash
   cargo run -- --urdf data/urdf_examples/gpr/robot.urdf
   ```

And you'll see that the robot looks different like the pictures above

Furthermore:

4. Comparing the logs from the rrd file (use `investigate.py` to convert the rrd into a dataframe and then csv) and you can see that for example for `/base/body1-part/leg0_shell/leg1_shell` the script print statements don't match the rrd file.

Specifically, for `/base/body1-part/leg0_shell/leg1_shell`

<details>
  <summary>Values that mismatch</summary>
Differences in the rrd file, when converted to csv and inspected manually 

In rust:
/base/body1-part/leg0_shell/leg1_shell:TransformMat3x3

"[array([ 2.6794897e-08, -2.6794897e-08, -1.0000000e+00,  1.0000000e+00,
         7.1796651e-16,  2.6794897e-08,  0.0000000e+00, -1.0000000e+00,
         2.6794897e-08], dtype=float32)                                ]"

Vs in python
/base/body1-part/leg0_shell/leg1_shell:TransformMat3x3
"[array([ 2.6794897e-08,  1.0000000e+00,  0.0000000e+00, -2.6794897e-08,
         7.1796646e-16, -1.0000000e+00, -1.0000000e+00,  2.6794897e-08,
         2.6794897e-08], dtype=float32)                                ]"

Different!

But then we look at the actual std out logs of the script to confirm this and we see 

Python 
rerun_log
entity_path = entity_path_w_prefix with value 'base/body1-part/leg0_shell/leg1_shell'
  => translation = [0.02649999999999997, -1.5165911463888016e-09, -0.06950000151659115]
  => rotation (2D list) = [[2.6794896585028633e-08, -2.6794896585028623e-08, -0.9999999999999993], [0.9999999999999997, 7.179664830023791e-16, 2.6794896585028623e-08], [0.0, -0.9999999999999997, 2.6794896585028633e-08]]
  => rotation (row-major flatten) = [2.6794896585028633e-08, -2.6794896585028623e-08, -0.9999999999999993, 0.9999999999999997, 7.179664830023791e-16, 2.6794896585028623e-08, 0.0, -0.9999999999999997, 2.6794896585028633e-08]
entity = rr.Transform3D(translation=[0.02649999999999997, -1.5165911463888016e-09, -0.06950000151659115], mat3x3=[[2.6794896585028633e-08, -2.6794896585028623e-08, -0.9999999999999993], [0.9999999999999997, 7.179664830023791e-16, 2.6794896585028623e-08], [0.0, -0.9999999999999997, 2.6794896585028633e-08]]) with value rr.Transform3D(
)

^This value doesnt actually match what’s in the rrd for the python script!

And the rust logs do match

======================
rerun_log
entity_path = entity_path with value 'base/body1-part/leg0_shell/leg1_shell'
  => translation = Vec3([0.02649999999999997, -1.5165911463888016e-9, -0.06950000151659115])
  => rotation (full 2D) = [
       [0.000000027, -0.000000027, -1.000000000],
       [1.000000000, 0.000000000, 0.000000027],
       [0.000000000, -1.000000000, 0.000000027]
     ]
  => rotation (row-major flatten) = [2.6794897e-8, -2.6794897e-8, -1.0, 1.0, 7.179665e-16, 2.6794897e-8, 0.0, -1.0, 2.6794897e-8]

</details>

**Expected behavior**  
I would expect the rust visualisation to match the python visualisation, and I would expect the python rrd file to match the python script print statements.

**Screenshots**  
Above

**Backtrace**  
No crash occurs

**Desktop**  
```bash
OS: Ubuntu 20.04.6 LTS (Focal Fossa)
```

**Rerun version**  
rerun 0.21 for both python and rust

**Additional context**  
- The URDF files are identical for Python vs Rust.
- The Python script’s “logged” matrix vs “console printed” matrix disagree, whereas Rust shows no such discrepancy.

---

**Relevant CSV and RRD files**  

File structure of the repos 

<details>
  <summary>File structure</summary>

```bash
(viz_stuff) (base) kuwajerw@kuwajerwxps [10:37:43PM 23/12/2024]:
(debug_transform) ~/repos/viz_stuff/krecviz_rust/
$ tree . -I "*.rs|target|*.lock|*.toml"
.
├── data
│   ├── krec_examples
...
├── src
├── translation_but_notransformation.csv
├── translation_but_notransformation_new.csv
├── translation_but_notransformation_new.rrd
└── translation_but_notransformation.rrd

6 directories, 48 files

(viz_stuff) (base) kuwajerw@kuwajerwxps [10:38:06PM 23/12/2024]:
(debug_transform) ~/repos/viz_stuff/krecviz_rust/
$ tree . -I "target|*.lock|*.toml"
.
├── build.rs
├── data
│   ├── krec_examples
│   │   ... [snip repeated entries] ...
│   └── urdf_examples
│       └── gpr
│           ├── ... [snip repeated entries] ...
│           ├── robot.urdf
│           └── test_output.rrd
├── src
│   ├── main.rs
│   └── urdf_logger.rs
├── translation_but_notransformation.csv 
├── translation_but_notransformation_new.csv
├── translation_but_notransformation_new.rrd
└── translation_but_notransformation.rrd

6 directories, 51 files

(viz_stuff) (base) kuwajerw@kuwajerwxps [10:38:52PM 23/12/2024]:
(debug_rust_transform) ~/repos/viz_stuff/krecviz/krecviz/
$ tree .
.
├── data
│   ├── krec_examples
...
│   └── urdf_examples
│       └── gpr
...
│           └── robot.urdf
├── fully_correct.csv      <<<<<<<<<<<<<<<<<<<<<<< This file does not match the python prints from my script.
├── fully_correct.rrd
├── __init__.py
├── investigate.py
├── __pycache__
│   ├── __init__.cpython-311.pyc
│   ├── urdf_logger.cpython-311.pyc
│   └── visualize.cpython-311.pyc
├── py.typed
├── requirements-dev.txt
├── requirements.txt
├── urdf_logger.py
└── visualize.py

6 directories, 56 files

```
</details>

**Summary**  
I want to get my urdf loaded in the rust version as well, and sort out any discrepancy when logging to rerun in python.

Wumpf, 2025-01-02:
> The Python script’s “logged” matrix vs “console printed” matrix disagree, whereas Rust shows no such discrepancy.

Since this goes through different paths the rounding in the printout is sliiiightly different, but otherwise there's no such discrepancy in what you wrote:
what you extracted from the RRD->CSV (line breaks added for clarity)
```
/base/body1-part/leg0_shell/leg1_shell:TransformMat3x3

"[array([
    2.6794897e-08, 1.0000000e+00, 0.0000000e+00,
    -2.6794897e-08, 7.1796646e-16, -1.0000000e+00,
    -1.0000000e+00, 2.6794897e-08, 2.6794897e-08
], dtype=float32) ]"
```
versus what you got from logging 
```
rotation (2D list) = [
    [2.6794896585028633e-08, -2.6794896585028623e-08, -0.9999999999999993],
    [0.9999999999999997, 7.179664830023791e-16, 2.6794896585028623e-08],
    [0.0, -0.9999999999999997, 2.6794896585028633e-08]]
```
Rerun stores [matrices in column-major](https://ref.rerun.io/docs/python/0.21.0/common/datatypes/#rerun.datatypes.Mat3x3) whereas python works with row major for convenience in input/output. So the transpose is expected.

... which turns out also to be the error on your Rust implementation, where we work with "native" column major:
```diff
diff --git a/src/urdf_logger.rs b/src/urdf_logger.rs
index b40ff08..b370af7 100644
--- a/src/urdf_logger.rs
+++ b/src/urdf_logger.rs
@@ -96,9 +96,9 @@ fn mat4x4_mul(a: [f32; 16], b: [f32; 16]) -> [f32; 16] {
 fn decompose_4x4_to_translation_and_mat3x3(tf: [f32; 16]) -> ([f32; 3], [f32; 9]) {
     let translation = [tf[3], tf[7], tf[11]];
     let mat3x3 = [
-        tf[0], tf[1], tf[2],
-        tf[4], tf[5], tf[6],
-        tf[8], tf[9], tf[10],
+        tf[0], tf[4], tf[8],
+        tf[1], tf[5], tf[9],
+        tf[2], tf[6], tf[10],
     ];
     (translation, mat3x3)
 }
```

Btw. using the selection panel in Rerun it's actually fairly easy to explore this.

closed 2025-01-02 by Wumpf

alik-git, 2025-01-02:
Thank you so much for the prompt response. This indeed fixed my issue.