rerun-io/rerun #8563: Use separate response messages in our gRPC API

emilk, 2024-12-26:
It is [best practice](https://kreya.app/blog/grpc-best-practices/#separate-request-and-response-messages) to use a different response message to each rpc call, as that allows us to add more fields to the response in the future without sacrificing compatibility.

That is, instead of 

```
    rpc QueryCatalog(QueryCatalogRequest) returns (stream DataframePart) {}
    rpc RegisterRecording(RegisterRecordingRequest) returns (DataframePart) {}
```

we should have

```
    rpc QueryCatalog(QueryCatalogRequest) returns (stream QueryCatalogResponse) {}
    rpc RegisterRecording(RegisterRecordingRequest) returns (RegisterRecordingResponse) {}
```

grtlr, 2025-01-03:
I can do this as part of #8566.

zehiko, 2025-01-07:
the most recent API changes were made as part of https://github.com/rerun-io/dataplatform/issues/74 (which is good to read for understanding the motivation behind the changes).

closed 2025-04-09 by teh-cmc