rerun-io/rerun #8660: Documented instructions do not work when using the C++ SDK in a shared library

svepe, 2025-01-13:
**Describe the bug**
Cannot build a shared library using the the C++ SDK by following the default instructions
since the static library is built without position independent code. 

**To Reproduce**
Steps to reproduce the behavior:
1. Follow these [C++ SDK instructions](https://ref.rerun.io/docs/cpp/stable/md__2home_2runner_2work_2rerun_2rerun_2rerun__cpp_2cmake__setup__in__detail.html#autotoc_md20) and add it to a shared library project
2. Attempt to compile
3. Results in an error since the static library was not compiled with the `-fPIC` flag on

**Fix**
I managed to get things working by compiling from source with the following flags:

```
cmake -B build -S . -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_FLAGS="-fpic"
```

I think it's a good idea to either add the flag by default (not sure what the repercussions are) or mention it in the documentation. Happy to look into either and submit a PR.

Wumpf, 2025-01-13:
Afaik enabling position dependent code always/when not needed can hurt performance (see e.g. [here](https://stackoverflow.com/questions/813980/why-isnt-all-code-compiled-position-independent) and [here](https://stackoverflow.com/questions/31332299/why-not-always-use-fpic-position-independent-code)), we don't want to turn it on always

I wonder if we could do something like this:
```
if(BUILD_SHARED_LIBS)
    set_property(TARGET rerun_sdk PROPERTY POSITION_INDEPENDENT_CODE ON)
endif()
```
in `rerun_cpp/CMakeLists.txt` (using CMake's [`POSITION_INDEPENDENT_CODE`](https://cmake.org/cmake/help/latest/prop_tgt/POSITION_INDEPENDENT_CODE.html#prop_tgt:POSITION_INDEPENDENT_CODE) option which is a bit more robust & clearer imho)

But [`BUILD_SHARED_LIBS`](https://cmake.org/cmake/help/latest/variable/BUILD_SHARED_LIBS.html) is a default setting so I'm not sure if that's the right thing to check for. Might work in some cases but not others.
Maybe there's a way to check whether the user requested `rerun_sdk` specifically as a shared library and then set the flag?

As you say maybe it's enough pointing this out in the docs - CMake documents this here as well btw. as a separate thing https://cmake.org/cmake/help/latest/guide/tutorial/Selecting%20Static%20or%20Shared%20Libraries.html

svepe, 2025-01-15:
Given the possible performance issues, then mentioning it in the documentation should be enough. There is also [this option](https://github.com/gabime/spdlog/issues/1190#issuecomment-650484907) for the case where FetchContent is used.