rerun-io/rerun #8662: Do not attempt to render events for streams that are not scrolled out of screen

abey79, 2025-01-13:
Querying and rendering the event view for streams in the time panel is, despite recent chunkification, quite slow. This is made worse by the "expand all" behaviour of the filter feature (https://github.com/rerun-io/rerun/pull/8654).

A likely optimisation is:
- to avoid rendering evens for everything that is above the visible part of the scroll area
- early out as soon as we're bellow the visible part of the scroll area

abey79, 2025-01-21:
Maybe something I can sneak in when I do the refactor this cycle.

emilk, 2025-01-21:
Is this a performance drain when there are many entities? What is the slow part of it? Do you have a profiler screenshot?

abey79, 2025-01-21:
The time panel is definitely a time hog when everything is uncollapsed. However, contrary to what I initially thought, we _already_ skip drawing the event when not in the visible area.

The "early-out-when-bottom-is reached" optimisation sound a bit stupid (things would get worse upon scrolling the list), it does make sense in the context of filter. Indeed, early during typing the query, everything that matches (which can be a lot) is shown uncollapsed. Early exiting would allow keeping the frame time reasonable during the early moment of the user typing the query.

### Collapse all

[image]

### Expand all

[image]

### Expand all profile

[image]

abey79, 2025-01-24:
This issue is related and likely relevant:
- https://github.com/rerun-io/rerun/issues/8798

closed 2025-01-29 by emilk