rerun-io/rerun #8665: BUILD.md doesn't list a specific pixi version, but the current pixi release is incompatible with rerun (apparently)

photex, 2025-01-13:
**Describe the bug**
Attempting to setup a development environment by following the instructions in BUILD.md (Linux) and at the step `pixi run check-env` the following is the result:

```sh
[chipc@bucket rerun]$ pixi run check-env
 WARN Encountered ambiguous version specifier `1.5.1`, could be `1.5.1.*` but assuming you meant `==1.5.1`. In the future this will result in an error.
 WARN Encountered ambiguous version specifier `18.0.0`, could be `18.0.0.*` but assuming you meant `==18.0.0`. In the future this will result in an error.
 WARN Encountered ambiguous version specifier `19.37.32822`, could be `19.37.32822.*` but assuming you meant `==19.37.32822`. In the future this will result in an error.
 WARN Encountered ambiguous version specifier `16.0.6`, could be `16.0.6.*` but assuming you meant `==16.0.6`. In the future this will result in an error.
 WARN Encountered ambiguous version specifier `1.11.1`, could be `1.11.1.*` but assuming you meant `==1.11.1`. In the future this will result in an error.
 WARN Encountered ambiguous version specifier `16.0.6`, could be `16.0.6.*` but assuming you meant `==16.0.6`. In the future this will result in an error.
 WARN Encountered ambiguous version specifier `3.27.6`, could be `3.27.6.*` but assuming you meant `==3.27.6`. In the future this will result in an error.
 WARN Encountered ambiguous version specifier `1.8.0`, could be `1.8.0.*` but assuming you meant `==1.8.0`. In the future this will result in an error.
 WARN Encountered ambiguous version specifier `1.11.1`, could be `1.11.1.*` but assuming you meant `==1.11.1`. In the future this will result in an error.
 WARN Encountered ambiguous version specifier `18.0.0`, could be `18.0.0.*` but assuming you meant `==18.0.0`. In the future this will result in an error.
 WARN Encountered ambiguous version specifier `0.3.5`, could be `0.3.5.*` but assuming you meant `==0.3.5`. In the future this will result in an error.
Error: 
  × Unexpected keys, expected only 'cmd', 'depends-on', 'description'
     ╭─[pixi.toml:214:9]
 213 │ ] }
 214 │ fmt = { depends_on = ["format"] }
     ·         ─────┬────
     ·              ╰── 'depends_on' was not expected here
 215 │ 
     ╰────
  help: Did you mean 'depends-on'?
```

**Expected behavior**
Not entirely sure since I'm new to this. But there is a mismatch between the documentation and the current versions of the software used to build rerun.

**Desktop (please complete the following information):**
Arch Linux, KDE 6.2

**Rerun version**
HEAD (~2025/01/13)

photex, 2025-01-13:
I see it mentioned here in fact: https://github.com/prefix-dev/pixi/releases/tag/v0.40.0

Wumpf, 2025-01-13:
thanks for reporting! Until now pixi has been quite stable with the pixi.toml file, but looks like they broke compatibility now and unfortunately it's not yet possible to specify a version in the pixi configuration

As of writing, we're still on pixi 0.39

photex, 2025-01-13:
Looks like it's an easy fix and resolution too. :D It was a toss up to decide whether to report this in each project because it seems like one of those things that will quickly resolve themselves, but I didn't want to let it go in case others run into it and didn't see an issue.

Wumpf, 2025-01-13:
Mentioned it over there now
* https://github.com/prefix-dev/pixi/issues/1945#issuecomment-2587323977

closed 2025-01-14 by Wumpf