rerun-io/rerun #8668: Large performance regression for missing query cache

Wumpf, 2025-01-13:
This test scene [alien_cake_addict.rrd.zip](https://github.com/user-attachments/files/18398780/alien_cake_addict.rrd.zip) plays (!) now a lot slower.

On my Mac Rerun reports 90ms on 0.21 and 270ms cpu time on 92949cf.
When paused it it went from 50ms to 65ms (also bad, but not as crazy).

From a first glimpse it seems that the cost of not hitting the query cache has gone up considerably, but more investigation is needed.
This may be temporary due to the [ongoing arrow/arrow2 conversion](https://github.com/rerun-io/rerun/issues/3741). (shipblocking in any case!).

emilk, 2025-01-13:
Since this is a scene with a huge number of objects, I suspect the added overhead of arrow1<->arrow2 conversions is to blame. The conversions are always zero-copy, but usually still involve a few small allocations and pointer-chasing, so they are by no means free.

emilk, 2025-01-14:
`fn row_sliced` is being called almost 9k times per frame, and is around 10µs slower than it was before (per call). I'm investigating why.

emilk, 2025-01-14:
This is super-weird. It is this piece of code that has become slower:

https://github.com/rerun-io/rerun/blob/0.21.0/crates/store/re_chunk/src/slice.rs#L110-L118

The things is - the code hasn’t changed:

https://github.com/rerun-io/rerun/blob/main/crates/store/re_chunk/src/slice.rs#L109-L117

Nor has the types surrounding it. Maybe the new Rustc optimizes worse? I think I need to `git bisect` :sob:

emilk, 2025-01-14:
This is the PR that regressed performance:
* https://github.com/rerun-io/rerun/pull/8617

But why?

closed 2025-01-14 by Wumpf