rerun-io/rerun #8693: There's NearClipPlane, but what about FarClipPlane for Spatial2DView???

ceubel-pr, 2025-01-15:
Having the ability to "crop" PCDs from a 2D particular view, based on some min and max distance, would be very valuable for the PythonSDK. It would allow us to take cross-sections of PCDs along some trajectory. There's already the NearClipPlane, adding the FarClipPlane would make this possible!

Posting NearClipPlane here for reference https://ref.rerun.io/docs/python/0.21.0/common/blueprint_components/#rerun.blueprint.components.NearClipPlane
and recent PR https://github.com/rerun-io/rerun/pull/8348

PS. setting clip planes directly in the 3D view and measurement features (like Meshlab) would also be nice :)

Wumpf, 2025-01-15:
We always render with [infinite reverse-z projection](https://developer.nvidia.com/blog/visualizing-depth-precision/), therefore we don't have a far clip plane to expose right now.
Clip planes on camera projections are often seen as an unwanted limitation, so I reckon since you want to use it for visualization you'd like them fixed in scene space rather than attached to a camera?

PCD==point cloud data?

Wumpf, 2025-01-15:
Very related (if not redundant to):
* https://github.com/rerun-io/rerun/issues/6247

ceubel-pr, 2025-01-15:
Ah, I see. Thank you.

I suppose I was thinking of adding virtual cameras to view 2D cross-sections of a point cloud, given a near and far plane. This is more of a post-processing, analysis step on top of the logged data. 

For instance, say I 3D scan a tree trunk, where each 3D scan is a logged timestep, but then I want to also view each cross-section of the composite tree trunk scan along the trunk's axis, say at 1" intervals to inspect it. My thought process is to create virtual cameras along the trunk axis and then control the near/far clipping, and then I would be able to flip through each of these cross-sectional waypoints to visualize. I'm trying to use rerun as a post-processing analysis tool on top of visualizing the scanning process, I am not sure if this is good practice. But, its been so powerful for the latter, it would be awesome to have features that support the former.

"PCD" really is a point cloud file format, but I too often subconciously use it to just mean a point cloud.

Wumpf, 2025-01-15:
Makes a lot of sense to me, thanks for the added context!
Might be challenging, but I think you're right this can be a powerful tool to have, definitely worth looking into

abey79, 2025-01-23:
This would be an awesome feature to have!

In the meantime, by using a combination of `rr.send_column()` and the `partition()` component method, you could log a point cloud with each slice assigned to a specific frame number of a dedicated frame timeline. That would require you to sort points by depths before logging and compute the partition indices (or, rather, the number of points belonging to each slice), but then it would be a rather powerful visualisation that is readily available with rerun:

- Scrubbing the timeline shows you a single slice with varying depth.
- Using visible time range, you can achieve various things such as thickening the slice (by using relative time delta), "growing" the point cloud ("from beginning to time cursor"), or display the whole thing ("from beginning to end")

Thanks to a bunch of work we did last year, this should come at no significant cost compared to just logging the entire point cloud in a single time point (except for the pre-processing on your end).

This example is very relevant about the above techniques: https://github.com/rerun-io/rerun/blob/main/examples/python/drone_lidar/drone_lidar.py