rerun-io/rerun #8712: Dataframe view handling of timeline is broken

abey79, 2025-01-16:
The timeline view appears to incorrectly behave when the timeline it's configured with does not exist. Unfortunately, that's a very common situation:

1) run some logging script with a custom timeline
2) configure a dataframe view set on that custom timeline
3) run the logging script again
4) 💥 

What seem to happen is that when the 2nd recording is created but before the first data reaches the viewer, the dataframe view doesn't see the timeline it's configured for and incorrectly handle the situation.

This can be reproduced with the simplest of the script:

```python
import rerun as rr

def main():
    rr.init("rerun_example_bug_dataframe")
    rr.connect_tcp()

    rr.set_time_sequence("another_timeline", 0)

    rr.log("pt", rr.Points3D([0, 0, 0]))

if __name__ == "__main__":
    main()
```

abey79, 2025-02-03:
Discord thread with relevant user workflow: https://discord.com/channels/1062300748202921994/1329143177134735432/1329143177134735432

valyagolev, 2025-02-05:
Thank you, looking forward to seeing this released

PR #8908 (2025-02-04): Support unknown timeline in dataframe view
### Related

- Fixes #8712 

### What

Initially, the dataframe view was built around the idea that the timeline it's configured for is always valid. It uphold this invariant, it would auto-switch to a known timeline in case the current on is unknown.

This was deeply flawed:
1) Recording has some stuff on timeline "X".
2) User configures a dataframe view with timeline "X"
3) New recording of the same app_id has _nothing_ on timeline "X" (e.g. because some event didn't happen in this run, or just because it takes time for data to be logged on "X")
==> [... cut]