rerun-io/rerun #8729: Unable to silence warning: Keypoint connection from index KeypointId(...) to KeypointId(...) could not be resolved in object [...]

sgiessl, 2025-01-17:
I want to use rerun with Points2D (or Points3D) and with connecting lines according to the respective AnnotationContext/ClassDescription's `keypoint_connections = ...` specification. This works well as long as the logged Points2D archetype defines all keypoints and -IDs which shall be connected.

In my use case, the logged Points2D are not always complete, missing points which are connected via `keypoint_connections`. The desired behavior is that the respective point and all of its connections are not displayed. This is in fact the actual behavior of the rerun viewer. But ...

**Describe the annoyance**

...The problem is that there is a warning, such as the following: `[2025-01-17T17:47:17Z WARN  re_view_spatial::visualizers] Keypoint connection from index KeypointId(0) to KeypointId(1) could not be resolved in object /test`

Such a warning on the terminal would not actually be a huge problem. The problem, however, is that the warning is also shown in the UI, as seen in the screenshot below. There may be multiple components of different path, thus in practice the UI gets cluttered by many repeating warning messages of the same kind, at different times in the rerun usage.

**To Reproduce**

Run the attached minimal sample program.

**Expected behavior**

Either:

* Make it only one single general warning of the same kind that's displayed once
* Don't log these errors in the UI
* Make the log level that's printed in the UI configurable, so users may exclude WARN level if they want to
* Or... **Don't treat it as a warning? It may be expected behavior in various scenarios. Think of human pose detections, for example, where not all keypoints may be visible at all times**

**Screenshots**
[image]

(see warning shown on the top right)

[rerun_min_example.py](https://github.com/user-attachments/files/18458235/rerun_min_example.py.txt)

teh-cmc, 2025-01-20:
Quick workaround: set the `RUST_LOG` environment variable, it controls the verbosity of both the viewer and the terminal, e.g. `RUST_LOG=error`.

sgiessl, 2025-01-20:
Thank you @teh-cmc, while trying various RERUN_LOG_LEVEL ideas, I didn't think of that obvious one. Although the rust log system could be mentioned somewhere in the docs, I'm sure it will show up immediately in future searches.

This is good enough for my needs, so to the person triaging: I'm leaving this open for your review, but fine by me to close. Feel free to reference my use case for "minor annoyances" backlog inspiration.

sgiessl, 2025-01-21:
Hi @teh-cmc sorry to get back with bad news. Turns out your proposed work-around actually does _not_ seem to work. I'm running `/usr/bin/python3 /home/user/.local/bin/rerun --hide-welcome-screen -j 8 --expect-data-soon --memory-limit 25% --drop-at-latency 750ms` and I've confirmed that all child process instances (enumerated via `pstree -p`) do seem to receive the correct environment variable `RUST_LOG=error` when checking `/proc/${PID}/environ`.

I know this copy of rerun was installed via `pip3 install rerun-sdk==0.21` and I seem to remember the docs mentioning that rerun will integrate into the respective language logging system (python in this case?). I will later go and see if rerun installed via cargo might behave differently (note, I do only have a very fuzzy concept of how the rerun core integrates into languages and logging runtimes).

Any additional tips and ideas are much are appreciated :)