rerun-io/rerun #8744: Unify `sorbet` and `rerun` arrow metadata

jleibs, 2025-01-20:
We currently have 2 different arrow-metadata encoding schemas: 
- In query API results we use:
  - `sorbet.path`, `sorbet.semantic_family`, `sorbet.logical_type`, `sorbet.semantic_type`
- In chunk transport we use:
  - `rerun.entity_path`, `rerun.archetype_name`, `rerun.archetype_field_name`, Name of column as component name

Keeping track of which encoding is required at different points in the pipeline is hard and only adds confusion while bringing no meaningful utility.

We should pick a single set of metadata keys and use them consistently as part of our first release of requiring tagged-component metadata to be populated.

jleibs, 2025-01-21:
Some frameworks have restrictions on what defines a valid column-name. For example: lance doesn't permit top-level field names to contain a ".".

This is a good argument for Rerun being agnostic about the choice of field-name and encoding all metadata as proper metadata fields.

emilk, 2025-01-28:
We discussed today a bit on how to handle _mono components_, i.e. the common case of single-instances (e.g. scalars). We want to make this as ergonomic as possible.

* We will start supporting RecordBatches with Mono-types in them, and do the list-array wrapping on the way into the chunkstore so that code in viewer-space never has to worry about mono-types.
* Future: we can also introduce a sorbet tag, e.g. `rerun.mono` and use that to unwrap-list arrays when generating query results in order to make these datatypes round-trip properly from the perspective of users.

emilk, 2025-02-05:
This is the direction I'm heading:

```rs
pub struct ChunkSchema {
    chunk_id: ChunkId,
    entity_path: EntityPath,
    columns: Vec<ComponentColumnDescriptor>,
    …
}
impl From<ChunkSchema> for ArrowSchema { … }
impl TryFrom<ArrowSchema> for ChunkSchema { … }

struct ChunkBatch {
    schema: ChunkSchema,
    batch: ArrowRecordBatch,
}
impl AsRef<ArrowRecordBatch> for ChunkBatch { … }
impl TryFrom<ArrowRecordBatch> for ChunkBatch { … }
```

And similar for dataframes:

``` rs
pub struct DataframeSchema {
    // Each data column is guaranteed to have an entity path
    columns: Vec<ComponentColumnDescriptor>,
    …
}
impl AsRef<ChunkSchema> for ArrowSchema { … }
impl TryFrom<ArrowSchema> for DataframeSchema { … }

struct DataframeBatch {
    schema: DataframeSchema,
    batch: ArrowRecordBatch,
}
impl Deref<Target = ArrowRecordBatch> for DataframeBatch { … }
impl TryFrom<ArrowRecordBatch> for DataframeBatch { … }
```

grtlr, 2025-02-07:
My comment on #8965 might be relevant here too:

https://github.com/rerun-io/rerun/pull/8965#discussion_r1946130488

> It would be really great to have that information as actual types. Conceptually, it would be great to have `struct Dataframe<I: Index>`, where `Index`, in this case, would be something like `ResourceId`. For data frames that don't have an explicit index (the order is the index), we could default to `struct Dataframe<I: Index = ()>`.
> 
> Of course `proto` does not give us generics, but maybe we could "monomorphize" the few instances where this is the case. We could then have specialized constructors + deserializers that ensure these invariants.

emilk, 2025-03-18:
What's still to be done is the dataframe portion of this. Should sync with @timsaucer here.

emilk, 2025-04-17:
Also some misc cleanup:
* [ ] `Chunk::to_chunk_batch` should be infallible