rerun-io/rerun #8776: Replace `StoreInfo` with "recording properties" stored as a `RecordBatch`

emilk, 2025-01-22:
We should have some standard columns in it, e.g.
* `rerun.date_created`
* `rerun.app_id`
* …

But otherwise let users add whatever they want to it.

The StorageNode should ingest this and put it in the catalog DB.

It would also be nice to support _appending_ data to the properties with a custom message in the .rrd, so we can add properties to an rrd by appending to it. This means replacing `LogMsg::SetStoreInfo` with something like `LogMsg::AppendProperties(StoreId, RecordBatch)`.

Exactly how the append works is TBD (e.g. if we have the same column name but different datatypes).

grtlr, 2025-02-27:
I think we even talked about having properties as static data without an entity path.

Both options should work. I think it would be more intuitive to not have that data be part of the entity tree, if we want to have a "recording properties" area somewhere _outside_ of the central panel (where normal views) go.

Either way, we should also make setting the properties a distinct call, because it be much easier to discover:
```rs
rec.set_properties(/* ... */); // Works with autocomplete
```
Rather than something like this:
```rs
rec.log("/", archetypes::RecordingProperties::new(/* ... */));
```

grtlr, 2025-02-28:
## Some notes

Current register endpoint:

* Takes `StoreInfo`, merges it with catalog metadata.

### Design:

* Properties stored in `.rrd` will never change, but we can add columns to it.
* Start moving fields from `StoreInfo` to `archetypes::RecordingProperties`
  * Still requires full scan
  * `RecordingId` should live outside of `archetypes::RecordingProperties`

### Problems:

* We can have streams that go to different stores (recording, blueprint).

grtlr, 2025-03-18:
The ground work for this issue was done in #9249.

The leftover work is tracked in #9190 with more information there, which is why I'm closing this issue.

closed 2025-03-18 by grtlr