rerun-io/rerun #8781: PythonAPI: Adding additional AnyValues with lists of strings sometimes give (empty) or truncated text

mnbvmnbv2, 2025-01-22:
Could not find similar bug

**Describe the bug**
In PythonAPI: When appending `rr.AnyValues` with a list of strings to `Boxes3D` the values often show as (Empty).
It is a bit inconsistent when it happens, but in my experience it happens mostly when the list is created dynamically and could contain for example `None` values (even when these are converted to string) or when the length of the strings vary.
I have had similar issues with this python API where floats was converted to ints unless I added a 0.0 at the end of the list for example.

[image]

**Workaround** I have found that appending one large string value could fix the problem. And depending on the length it could truncate the other strings if it is shorter. 
Example:

[image]
From code:

[image]

**To Reproduce**
Steps to reproduce the behavior:

Something like this:

[image]

**Expected behavior**
I expected the values to show in the rerun viewer like this:

[image]

**Backtrace**
No trace

**Desktop (please complete the following information):**
 - Ubuntu 22.04.4 LTS

**Rerun version**
 0.20.0 and 0.21.0: Python API

**Additional context**
I cannot give my own code, but could try to give a fully reproducible code if there is interest in solving the issue.
My guess is that it is some typing issue (specifically with characters/string and floats) when the python code is sent to Rust.

mnbvmnbv2, 2025-01-22:
Also "labels" in Boxes3D does not seem to have this problem so another "workaround" is to move the values there...

brookman, 2025-02-21:
We ran into the same issue with the Python SDK.

After doing this
```python
 rr.log("/world/data", rr.AnyValues(MyString="abc"))      
```
the "MyString" cloumn somehow seems to be fixed to strings of length 3.
When doing this later:

```python
 rr.log("/world/data", rr.AnyValues(MyString="1234"))      
```
Rerun will silently truncate the string "1234" to "123".

timsaucer, 2025-03-12:
The root of this problem appears to be due to the calls in `rerun_py/rerun_sdk/rerun/any_value.py` where we convert the values to numpy arrays before turning into pyarrow arrays. After the first call we use the returned type to cast subsequent values. When `np.asarray()` is called for the first string it returns a fixed length string.

I will work on this issue. My plan is to see if the value is an iterable or numpy array. If so attempt to turn them directly into pyarrow arrays. If not, attempt a cast to pyarrow scalar and turn it into a pyarrow array of dimension 1.

For the reporters, a work around that I have tested is to pass a pyarrow array of string type.

In the second user example, you should be able to use this work around:

```python
 rr.log("/world/data", rr.AnyValues(MyString=pa.array(["abc"])))      
```

PR #9269 (2025-03-14): Correct truncation of AnyValues when using strings or bytes
### Related

* Closes #8781

### What

This PR adjusts the way we process AnyValues. WIthout this change we have cases where numpy is setting data types to have fixed length based on the first value it receives. This ends up truncating the data.

Additionally, we cannot simply call `pa.array()` because types like strings and bytes are iterable and will get turned into an array of characters or bytes, respectively.

This PR attempts three passes at converting values

- Attempt to call `pa.array()` directly, but with a special case to i [... cut]
commit 3ba55c51 (2025-03-14): Correct truncation of AnyValues when using strings or bytes (#9269)

### Related

* Closes #8781

### What

This PR adjusts the way we process AnyValues. WIthout this change we
have cases where numpy is setting data types to have fixed length based
on the first value it receives. This ends up truncating the data.

Additionally, we cannot simply call `pa.array()` because types like
strings and bytes are iterable and will get turned into an array of
characters or bytes, respectively.

This PR attempts three passes at converting values

- Attempt to call `pa.array()` directly, but with a special  [... cut]