rerun-io/rerun #8811: Implement screen space ambient occlusion (SSAO) for 3d viewer

jviereck, 2025-01-26:
See conversation about implementing lights and shadows here: https://github.com/rerun-io/rerun/issues/4845

jviereck, 2025-01-26:
Looking for references on how to implement SSAO, this one looks good to me: https://learnopengl.com/Advanced-Lighting/SSAO

@Wumpf , do you think this way of implementing SSAO is good?

Wumpf, 2025-01-27:
Don't have much time right now to go through & evaluate SSAO tutorials ;-) but this one looks like it needs a normal backbuffer and we don't have that right now (and I don't think we'd go with a deferred renderer any time soon, so it remains scarce), this can be generated from depth ofc but I'd rather go with a technique that doesn't have this requirement. There's tons of different SSAO techniques but I'm fairly sure I saw survey papers popping up in recent years, that's where I'd start.

jviereck, 2025-01-27:
Hi @Wumpf , no worries - I don't expect you to go in details through tutorials or look at them. The high level comment about missing normal backbuffer was super helpful. I will look for a different implementation then.