rerun-io/rerun #8814: Display Poses on MapView

LoyVanBeek, 2025-01-27:
Related to https://github.com/rerun-io/rerun/issues/8087 in a way

**Is your feature request related to a problem? Please describe.**
I would like to show the direction a robot is driving or which way a camera is facing on a map. Currently, Rerun can only show a `GeoPoint`, but no associated heading, no less a full 3D orientation.

**Describe the solution you'd like**
Besides `GeoPoint`, add eg. a `GeoPose` datatype that includes a heading, so that an arrow can be drawing pointing in a given direction. 

For drone applications, an extension of this would be to also include a full 3D orientation as well as altitude. 

To take this to the full extent, ideally Rerun can show a full 3D view (like that in the NuScenes example) on top of a MapView.

**Describe alternatives you've considered**
A workaround is drawing an arrow or camera frustum using a `GeoLinestring`.

**Additional context**
My project involves a robot to accurately geotag images of crops, so that after processing the images for signs of mold and other diseases etc, the plants can be found again by a human or robot and treated.
For this, I'd like to visualize these images on a map view, eg some combination of the [NuScenes example](https://rerun.io/examples/robotics/nuscenes_dataset) wit the camera view overlayed on a 3D rendition of the map. We're in the Netherlands, 3D sloped terrain rendering is not a priority 😄

abey79, 2025-01-27:
Thanks for the write-up.

Adding `GeoArrows` (*) as a map sibling to [`Arrows2D`](https://rerun.io/docs/reference/types/archetypes/arrows2d) seems like a reasonably well defined improvement that wouldn't be overly complicated. With `heading` and `length` (in m) instead of `vectors`, such an archetype definition would resist future flexibility in the projection used by the map view.

The more generalised "project any 3D object on the map" is tricker. The right way forward would be to devise a general mechanism to assign localisation information to any existing archetype, such that it could the meaningfully represented on a map. That might look like an extension of the existing transform system. This is certainly something we'd like to eventually have, but the timeline line for that is very uncertain.

(*) somewhat [unfortunate](https://geoarrow.org) name but unsure what else it could be

LoyVanBeek, 2025-01-27:
How about `GeoPose2D` or `GeoArrows2D`?

LoyVanBeek, 2025-01-28:
Ha, shower thought: I could download map tiles in my application and display those via eg. a `Pinhole` camera into a 3D view.

abey79, 2025-01-28:
> Ha, shower thought: I could download map tiles in my application and display those via eg. a `Pinhole` camera into a 3D view.

You can also use the map tiles as texture on a 3D mesh in a 3D view.

LockedThread, 2025-07-22:
> could download map tiles in my

Did you ever do this?

LoyVanBeek, 2025-07-22:
> > could download map tiles in my
> 
> Did you ever do this?

No, haven't tried either, no time.