rerun-io/rerun #8815: Make the `ListItem` offscreen optimisation opt-out after 0.22

abey79, 2025-01-27:
This PR introduces an opt-in offscreen optimisation for ListItem:
- https://github.com/rerun-io/rerun/pull/8808

Let's make that opt-out right after the next release, to be on the  safe side.

nikolausWest, 2025-01-27:
Why not include it?

abey79, 2025-01-27:
When that PR lands (which is as soon as the other PR it's chained to lands), then that optimisation will be active where it maters (and where I've properly tested it): blueprint tree and streams tree. We're just postponing planet activation just to be on the safe side (and also because there is is next to nothing to be gained with it in most places where lists are short).

abey79, 2025-08-29:
this is already done

closed 2025-08-29 by abey79