rerun-io/rerun #8841: Panic originating from sorting function.

02alexander, 2025-01-28:
The rerun viewer sometimes crashes with the following error:
```
thread 'main' panicked at 'user-provided comparison function does not correctly implement a total order'
core/src/slice/sort/shared/smallsort.rs:865
stack backtrace:
   6: core::panicking::panic_fmt
   7: core::slice::sort::shared::smallsort::panic_on_ord_violation
   8: core::slice::sort::shared::smallsort::bidirectional_merge
   9: core::slice::sort::shared::smallsort::small_sort_general_with_scratch
  10: core::slice::sort::stable::quicksort::quicksort
  11: core::slice::sort::stable::drift::sort
  12: core::slice::sort::stable::driftsort_main
  13: egui::context::Context::begin_pass
  14: egui::context::Context::run
  15: eframe::native::epi_integration::EpiIntegration::update
  16: <eframe::native::wgpu_integration::WgpuWinitApp as eframe::native::winit_integration::WinitApp>::run_ui_and_paint
  17: eframe::native::event_loop_context::with_event_loop_context
  18: <eframe::native::run::WinitAppWrapper<T> as winit::application::ApplicationHandler<eframe::native::winit_integration::UserEvent>>::window_event
  19: core::ops::function::impls::<impl core::ops::function::FnMut<A> for &mut F>::call_mut
  20: winit::platform_impl::linux::x11::EventLoop<T>::run_on_demand
  21: eframe::native::run::run_wgpu
  22: eframe::run_native

```
It happened while I was looking around in a 3D view, but I'm not able to reproduce it reliably.

**Desktop:**
 - OS: NixOS with KDE Plasma desktop.

**Rerun version**
```
rerun-cli 0.21.0 (default map_view native_viewer web_viewer) [rustc 1.83.0 (90b35a623 2024-11-26), LLVM 19.1.1] x86_64-unknown-linux-gnu
Video features: av1 default ffmpeg serde
```

02alexander, 2025-01-29:
It has happened 6 more times today. This problem has only appeared since I updated to 0.21.0 from 0.20.3 this morning.

Wumpf, 2025-01-29:
Duplicate of:
* https://github.com/rerun-io/rerun/issues/8530

This is a known issue in 0.21 when compiling the viewer from source with using rustc 1.81 or newer. The pre-builts don't have this issue and it is already fixed on `main`

closed 2025-01-29 by Wumpf