rerun-io/rerun #8844: Lineseries in the same TimeSeries plot have the same colors

jviereck, 2025-01-28:
In my setup I let rerun select the colors of lines. Here is an example where rerun chooses the exact same colors for two lines in the same TimeSeries plot. Having the same colors makes it very hard to distinguish the two plotted lines.

Rerun should assign unique colors to lines in the same TimeSeries plot if possible.

[image]

Wumpf, 2025-01-29:
We generate those default colors from hashing the entity paths. Hash collision would be very unlikely and in fact these are _not_ the same color

[image]
Top is `4dbca2`, bottom is `4eb9bc` in html notation. Aka: It's terribly subtle and very hard to tell!

Goes to say that our color generation approach isn't all that great. We need to move to sampling from a fixed color palette instead and use a different criterium!

jviereck, 2025-01-29:
@Wumpf thank you for your input! 

This page is very good for color palettes: https://seaborn.pydata.org/tutorial/color_palettes.html

jviereck, 2025-01-29:
@Wumpf , could you give me a pointer where the current colors are decided in the code? I might take a look if there is a easy way to improve the current situation.

Wumpf, 2025-01-29:
I think almost all of those end up eventually in here
https://github.com/rerun-io/rerun/blob/46bca8859d77647b94a262ca5b33e6f34e9af89f/crates/viewer/re_viewer_context/src/utils/color.rs#L4

Wumpf, 2025-01-29:
Imho the main issue is less the color palette (although that could be better as well!) but rather the fact that we can't easily make this depending on "surrounding state" today: any approach that has to pick from an "infinite" palette of colors using an effectively random number may run into the issue since any two numbers may produce colors that are too close to each other.
If we'd instead statefully take colors from a list rather than freely picking this would work a lot better. But that's ofc then a major change with a lots of complications.

jviereck, 2025-01-29:
> Still worth playing around with this oracle code though I figure 🤷

Can you point me to the oracle code?

jviereck, 2025-01-29:
> If we'd instead statefully take colors from a list rather than freely picking this would work a lot better.

I find a predefined list also helpful to figure out which line is the first, second etc. I know the matplotlib color by hard, so if I see the first listed item in the legend, I know the color automatically.

Wumpf, 2025-01-29:
> Can you point me to the oracle code?

well it's the same file :) - it just takes an entity path hash. You can use symbol search to check all the places that use that.

Wumpf, 2025-01-29:
sorry for the inprecise expression, I was referring to the tiny snippet I linked that gives a color from an integer.
Where those are produced is in the same file, see `auto_color_for_entity_path`. You can use symbol search to follow on where that's used in turn (lotsa places!)

jviereck, 2025-11-27:
hi @Wumpf, is there any update on this? How complicated to you think is this to attempt to fix myself?

Wumpf, 2025-11-27:
There's no api to check what's currently active in the viewer if that's what you mean. You could send new blueprints (or simpler just data) alongside that sets colors for everything whenever somethings new comes in independent of whether you'll actually show it. But I agree this  is neither great nor ergonomic since a) blueprint overrides are set on views so you have to populate a view with everything and then essentially set everything to invisible and b) every time you send a blueprint it overrides everything that you've configured in the viewer so far :/

>  I can assign colors to them from a fixed 10-color palette via python. [...]  So as the set of colors is fixed, this would still lead to color colisions.

That part I don't follow: whatever change we make to the viewer's fallback colors has to solve the same problem. Colors there need to be picked in a stable manner depending on as few criteria as possible (right now only the entity path goes into it) so that they don't change at random because of new data or some interaction

jviereck, 2025-11-28:
I was imagining the colors are asigned based on what lines are visible on the screen. First line displayed gets the first color from the palette and so on. This what matplotlib etc do. I feel this is different from what you use at the moment though.

Wumpf, 2025-11-28:
I don't think that's the right behavior for Rerun since I believe having colors to be consistent across views, even for different visualizations, is very useful 🤔. Which is something that matplotlib obviously doesn't have to care about

For example the arkit example uses default colors, but has automatically the same color across views for the same entities

[image]
(changed boxes in 3d to solid for illustrative purposes)

Also, what I really don't like about a simple index-based color assignment is that removing or adding an object from a view means that all other entities in the view may change their color, making it hard to keep track of things. Why would adding change it? There's no order to which things are added (_everything_ in Rerun can arrive out of order) so we'd have to rely on some other source of ordering like alphabetical ordering.
So once a color was decided it should no longer change, means we have to do some book-keeping.

Maybe there's a way to lazily assign colors to entities the first time they show up in any visualization, but I don't think this would hold up when closing & re-opening the viewer as the same data may again enter in an arbitrary order making it _really_ hard to get deterministic colors!

But maybe I'm generalizing too far here and we could just do something more specialized just for plot lines & points :). But the order stability problem would still be the same 🤔