rerun-io/rerun #8859: Add option for transform arrow visualizer behavior

Wumpf, 2025-01-29:
Historically, we too often had the problem that transform arrows would show up too eagerly and clutter the view a lot which is why we cut down on it over time. Last round of that was in 0.21 (see https://rerun.io/docs/reference/migration/migration-0-21#3d-transform-arrow-visualization-show-up-less-often-by-default): arrows are visualized by default iff

- enabled visualizer via VisualizerOverrides or ui
- AxisLength component is present as well
-  there's a pinhole camera at the same path

This works out fine in most cases but some users would like to see them more frequently. We should add an enum property to the 3D view to govern this behavior. Suggested possible option (names needed!):
* what we have right now
* everywhere where's a transform logged
* everywhere where's a transform logged and no other visualizer is active

More context in https://discord.com/channels/1062300748202921994/1334202640128802898/1334202640128802898

PeterMitrano, 2025-03-06:
I'm working on implementing something for this and could use some help!
I've started by creating `crates/store/re_types/definitions/rerun/blueprint/archetypes/transform_3d_arrows_blueprint.fbs` and then adding the UI to the 3d view using `view_property_ui::<Transform3dArrowsBlueprint>(ctx, ui, view_id, self, state);`. You can see the work in progress [here](https://github.com/rerun-io/rerun/compare/main...PeterMitrano:transforms3d_visualizer_props?expand=1).
Is this the right approach?

I also know where the heuristics are for the current logic around showing arrows, but I don't see how to access any of the blueprint state from that point in the code. Perhaps doing things inside `choose_default_visualizers` isn't the right approach anymore?

I would appreciate some any advice, thanks!

Wumpf, 2025-03-06:
Neat! The biggest issue you have there though is that right now we still can't do the same component type more than once on an single archetype. We already worked towards that a bit by recording the originating archetype & field name for each field, but as of now the viewer is completely oblivious to that information. In other words: if you write any of these fields, the value you get on any of the other ones is practically undefined behavior (for more details see https://github.com/rerun-io/rerun/issues/6889 unfortunately not entirely up to date)
Using several booleans as you tried here makes things more composable, but would today lead to small explosion of new component types, so I'd rather try to model it as an enum (which can be shown as a combobox then by adding a line in `re_component_ui`)

> I also know where the heuristics are for the current logic around showing arrows, but I don't see how to access any of the blueprint state from that point in the code. Perhaps doing things inside choose_default_visualizers isn't the right approach anymore?

I think we need to find a way to pipe the necessary information there. Haven't look at into it much, but shouldn't be too hard to get a `ViewContext` 🤔

What I'm almost more worried here is that since this method is called for every entity in the view, it could quickly become a performance issue to query the heuristic configuration for each call. Could use the `ViewState` to "cache" that I figure, just a little bit to update that then since that would need to happen in the view's `ui` method. But probably acceptable
(or maybe `visualizable_filter_context` can be extended for purposes like that? 🤔)