rerun-io/rerun #8866: Linking Objects Between 3D and 2D Views

ChengHsiuChi, 2025-01-30:
**Is your feature request related to a problem? Please describe.**
I’d like to double-click an object in the 3D view and have the corresponding object(s) shown in the 2D view.

**Describe the solution you'd like**
Add a link between objects. When we double-click an object, it makes the other one (in)visible.

**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**

**Describe alternatives you've considered**

**Additional context**

Wumpf, 2025-01-30:
Probably duplicated to
* https://github.com/rerun-io/rerun/issues/4379

But before I close this prematurely as duplicate, first like to understand the visibility ask here: You imagine strict toggling of related visualizations? So far we were thinking more in terms of highlighting.

Note that we already have some weak support for this: If the _same_ entity shows up in several views it will highlight accordingly. However, since right now different archetypes overwrite each other (see https://github.com/rerun-io/rerun/issues/6889) this is a bit limited since a single entity can rarely take over several roles in different views.

ChengHsiuChi, 2025-01-30:
Thank you for your prompt response!

The concept is similar to #4379 but with a few potential differences:

- The linked entities don't need to appear simultaneously.
- Only the selection action triggers the other entity.
- (Potential) Other entities turn invisible if not selected.

**Example**
[image]
In Spatial3DView, entities are linked to ones in the TimeSeriesView, and double-clicking an entity in the Spatial3DView triggers the corresponding entity to appear (solely) in the TimeSeriesView.