rerun-io/rerun #8876: Define and implement gRPC URI structure

emilk, 2025-01-31:
We want to be able to hand out URIs pointing at speicific things inside a storage node.
We also want to distinguish a link to a storage nide from a `re_grpc_server` (see https://github.com/rerun-io/rerun/issues/8761).

This is not part of the gRPC specification, so we need to define this ourselves.

## URI
Here's a reminder of what the different parts of a URI are called (from [wikipedia](https://en.wikipedia.org/wiki/Uniform_Resource_Identifier)):

[image]

### scheme
I think we should use `rerun://` as the scheme, to indicate that is this is something Rerun-specific. This makes it obvious what the URL is about, and we can potentially tell the OS to open such links using the `rerun` binary.

### Host/port
If no port is specified, we should use the default Rerun port.

### Path
Here are some suggestions:

* `/stream`: for connecting to an event stream. This is currently the only mode supprorted by `re_grpc_server`: forwarding all events from a SDK sink
* `/catalog/ID`: display a specific catalog
* `/store/ID`: download a specific recording and/or blueprint (or should we maybe distinguish between the two in the URI?)

#### Recording queries
When downloading a recording, we should be a ble to set specific query, to only download part of the recording. Most importantly, we want to filter on _time_, to either only download the data for a specific time point or a time range.

Suggestion:

* `?log_time=$TIME`: specific time point
* `?log_time=$TIME...$TIME`: time range (inclusive)

Where `$TIME` is one of the following:
* Sequence number on the format `frame_nr=#51` 
* Date-time in ISO 8601 (e.g. `log_time=2025-01-31T02:58:40Z`)
* Time offset/duration in seconds (`sim_time=120.12s`)

The viewer should download only the chunks necessary to show the given timepoint or time range, and then also move the time cursor to match.

## Open questions
* How do we distinguish between HTTP and HTTPS? Do we need to?

## Related issues
* https://github.com/rerun-io/rerun/issues/8761

emilk, 2025-02-07:
We should also have `/v0/` at the start of the path

jleibs, 2025-02-18:
Proposal for message-proxy using `rerun://` URIs (for consistency)
- `rerun://<address:port>/proxy` should connect to the message-proxy grpc interface.

grtlr, 2025-03-04:
I think this can be closed for now. Last related PRs:

* #9129
* #9157

closed 2025-03-04 by grtlr

jleibs, 2025-03-04:
I was still sort of hoping that this would result in a document with a formal spec mapping specific URI patterns to defined gRPC end-points in a deterministic way.

I realize that the fact that the grpc spec isn't finished precludes that from being done, and so maybe this scaffolding is enough and we should make a new issue for the formalization part.