rerun-io/rerun #9024: Logging level too low when running `rerun-cli --serve-web`

jleibs, 2025-02-12:
0.21:
```
➜  rerun --serve-web
[2025-02-12T18:33:49Z INFO  re_sdk_comms::server] Hosting a SDK server over TCP at 0.0.0.0:9876. Connect with the Rerun logging SDK.
[2025-02-12T18:33:49Z INFO  re_ws_comms::server] Hosting a WebSocket server on ws://localhost:9877. You can connect to this with a native viewer (`rerun ws://localhost:9877`) or the web viewer (with `?url=ws://localhost:9877`).
[2025-02-12T18:33:49Z INFO  re_sdk::web_viewer] Hosting a web-viewer at http://localhost:9090?url=ws://localhost:9877
```

0.22:
```
➜  rerun --serve-web

```
Having the INFO log statements with port number, etc are extremely important in this case.

Wumpf, 2025-02-12:
this was a deliberate change we made to reduce logspam in the context of the _viewer_. We have to find a compromise of when to show how much log

jleibs, 2025-02-12:
Yeah, the trade-off is hard.

That said, the real motivation for the change (https://github.com/rerun-io/rerun/issues/7815) wasn't to reduce logspam in the viewer, but logspam via the SDK when using in a notebook context.

I don't actually see an issue with the viewer sticking with INFO by default. It's the SDK info messages that are kind of spammy.

jleibs, 2025-03-06:
> We need to find a way to not have them show up in the notebook then
The notebook doesn't spawn a viewer instance.

Wumpf, 2025-03-06:
Sure those particular messages won't show up there, but the logspam issues was in particular about the notebook showing too much log from the viewer independent of what that shows in particular

emilk, 2025-03-07:
The cause of this problem is:
* https://github.com/rerun-io/rerun/pull/8918
*

Wumpf, 2025-03-07:
@jleibs and me had a brief call about this. Conclusions
* standalone cli should always show info
* sdk should show warning
* viewer opened from sdk is where things get trickier
    * if we're spawning a separate process (non-notebook) then probably default to what cli does, but we could pick what we want here explicitely
    * if we're running in the same process, do what sdk does (notebook!)

closed 2025-03-11 by Wumpf