rerun-io/rerun #9046: Add `EncodedDepthImage` archetype

oxkitsune, 2025-02-14:
`DepthImage`s are currently handled as raw `ImageBuffer` inside, but there are use cases where depth images are stored in encoded formats, such as TIFF (e.g., LeRobot datasets).

Since TIFF decoding is implemented in #8970, this functionality can be reused to support encoded DepthImages.

pablovela5620, 2025-02-18:
would something like this make it easier to compress depth maps so they don't take up as much memory (like rr.Image.compress() does)

oxkitsune, 2025-11-12:
> I have made a prototype working with re_mcap to visualize RVL encoded depth image from ros2 mcap bag.

That is seriously awesome! Looking forward to it!

> I would like to get advice from rerun team @oxkitsune on how to proceed merging this new feature into the codebase.

I think your suggestion of opening a draft PR is a good step, but I'd say that you can open the PR with the new archetype directly alongside it.

The other features (`re_mcap` etc) should be follow up PRs, agreed.