rerun-io/rerun #9078: Blueprint not preserved on webviewer

pablovela5620, 2025-02-18:
**Describe the bug**

Using this rrd https://huggingface.co/datasets/pablovela5620/rrd-examples/resolve/main/assembly-example.rrd
with the hosted webviewer here 
https://rerun.io/viewer?url=https://huggingface.co/datasets/pablovela5620/rrd-examples/resolve/main/assembly-example.rrd

does not preserve the saved blueprint. If one downloads the rrd and uses the native viewer, blueprint is correctly loaded

**To Reproduce**
Steps to reproduce the behavior:
1. go to https://rerun.io/viewer?url=https://huggingface.co/datasets/pablovela5620/rrd-examples/resolve/main/assembly-example.rrd
2. download https://huggingface.co/datasets/pablovela5620/rrd-examples/resolve/main/assembly-example.rrd
3. on local device run `rerun assembly-example.rrd

**Expected behavior**

The webviewer should show the same layout/blueprint as the native viewer

**Screenshots**

## Native layout
[image]

## Webviewer layout
[image]

**Rerun version**

```
rerun-cli 0.22.0-rc.1 (default grpc map_view nasm native_viewer release web_viewer) [rustc 1.81.0 (eeb90cda1 2024-09-04), LLVM 18.1.7] aarch64-apple-darwin, built 2025-02-06T19:56:15Z
Video features: av1 default ffmpeg nasm serde
```

teh-cmc, 2025-02-19:
That sounds like a duplicate of https://github.com/rerun-io/rerun/issues/8741, which will be fixed in either 0.22.1 (if there's such a thing), or 0.23.0.

Please try it on `main` (https://app.rerun.io/version/main/index.html?url=https://huggingface.co/datasets/pablovela5620/rrd-examples/resolve/main/assembly-example.rrd), but you'll need to regenerate the RRD first (`main` is on the new protobuf file format).

closed 2025-03-19 by Wumpf

pablovela5620, 2025-03-19:
Looks like that was exactly what was happening! Seems to be working great now
https://app.rerun.io/version/0.22.1/index.html?url=https://huggingface.co/datasets/pablovela5620/rrd-examples/resolve/main/hocap-example%2Brbl.rrd