rerun-io/rerun #9084: Uniquely define an index only by name

emilk, 2025-02-19:
We currently distinguish what we call _timelines_ by its name _and_ it's type (sequence or timestamp). We should instead only distinguish it by its name, and encode it's type only as the arrow datatype.

All data logged under a specific index column name must follow the same arrow datatype.

Advantages:
* Simpler
* Expand the possible types for index columns

## Related
* https://github.com/rerun-io/rerun/issues/8635

jleibs, 2025-02-19:
How do we intend to handle/surface this when two sources trying to write to the same recording disagree on their time type?

This is a tricky one to do gracefully while also handling whatever users throw at us.

emilk, 2025-02-19:
I think whenever you use different datatypes for the same column, there is no good way to handle it except log a warning or return an error. I don't think we should engineer something complex to handle something that users _shouldn't_ do.

jleibs, 2025-02-19:
> I don't think we should engineer something complex to handle something that users shouldn't do.

I sympathize with the perspective, but this seems like a potentially frustrating foot-gun we are leaving for users -- I want to make sure we are spending at least a little time documenting what the expectations are.

As it stands now you might not even see the error until some downstream process receives the chunk and tries to merge it into an existing (incompatible) store.

Maybe that's a server-backed store and no viewer is connected to it at the moment. Where does the error even go? If this is going to a grpc end-point maybe we can detect the problemon the server and back-channel an error and make this a fatal error in strict mode?

If we're not in strict mode, what happens to the data? Do we just throw it away?

Are there other tools we can at least plan for to make it easier for users to be aware of and not do the thing they shouldn't do?

emilk, 2025-03-27:
I  believe I completed all the important steps of this

closed 2025-03-27 by emilk